PlotJuggler/PlotJuggler #121: Load Layout via Launch file

djumpstre, 2018-12-04:
Hi, Davide, 

many thanks for that powerful tool. It's pretty useful! 

I have a simple question about the launch file, how can PlotJuggler load a predefined Layout via Launch file automatically? like: 

<node name = "PlotJuggler" pkg = "plotjuggler" type = "PlotJuggler"
          required = "true" output = "screen"
          args = "-d $(find package_name)/config/plot_juggler_test.xml"/>

Which key-word should used here instead of "-d" ? 

Thanks a lot, 
Dennis

awesomebytes, 2018-12-21:
Hello. As I was dropping by...
```bash
$ rosrun plotjuggler PlotJuggler --help
Usage: /opt/ros/kinetic/lib/plotjuggler/PlotJuggler [options]
PlotJuggler: the time series visualization tool that you deserve 

Options:
  -v, --version            Displays version information.
  -h, --help               Displays this help.
  -n, --nosplash           Don't display the splashscreen
  -t, --test               Generate test curves at startup
  -d, --datafile <file>    Load a file containing data
  -l, --layout <file>      Load a file containing the layout configuration
  --buffer_size <seconds>  Change the maximum size of the streaming buffer
                           (minimum: 10 default: 60)
```
`--layout` allows you to provide a layout file.

Given that you can just make a launchfile like:
```xml
<launch>
    <node name="plotjuggler_with_layout" 
          pkg="plotjuggler"
          type="PlotJuggler"
          args="--layout $(find my_package)/config/my_plotjuggler_layout.xml"
    />
</launch>
```
I just tried and it works.

facontidavide, 2019-01-08:
Thank you @awesomebytes

closed 2019-01-08 by facontidavide