PlotJuggler/PlotJuggler #122: Remove selected curves removes only clicked items

jaredkhan, 2018-12-20:
Environment: Ubuntu 16.04
To reproduce:
- open plotjuggler with the dummy data
- drag many series onto the plot
- right click on the legend
- click 'remove curves'
- click and drag over the list of items to select them all
- click 'remove selected'
Expected:
- All highlighted rows will be removed
Actual:
- Only rows that have been clicked, resulting in the name being struck through, will be removed

I really like the click and drag selection, and we get this for free from qt.

I propose we forgo the strike through and just when 'remove selected' is clicked, find the items that are selected and remove them (rather than finding the ones that have strikethrough font).

I propose marking as 'help wanted' since it would be a good starter bug

facontidavide, 2018-12-20:
Thank you very much for reporting. I will look into it

facontidavide, 2019-01-08:
Fixed. Thank you for reporting, it was really ugly

closed 2019-01-08 by facontidavide