PlotJuggler/PlotJuggler #129: Do not ask to load previous datafile, when -l and -d is specified

bysteven, 2019-01-25:
Thanks for contributing to PlotJuggler. You are great!

Please allow me to better understand your problem providing the following information:

- Please use always the master branch or the latest tag, if you can. 

- Check if the problem can be reproduced using the dummy data created by the command line argument "-t" or one of the files in the folder "datasamples".

- If it can't be reproduced with the dummy data, __please__ share the CSV file or the rosbag that can be used to reproduce the problem.

[image]

bysteven, 2019-01-25:
When I run plotjuggler and specify a layout as well as a datafile with -l and -d, respectively, it would be nice if it would not ask me whether I wanted to use the previous datafile.

facontidavide, 2019-01-25:
I also received requests from other users mentioning that they want to reuse the same layout with different datafiles... asking felt like the right thing to do. but I understand your point.

I will think about it.

facontidavide, 2019-01-25:
Let me know if that works for you

bysteven, 2019-01-27:
Very nice, thank you! I will give it a try and let you know. Cheers!