PlotJuggler/PlotJuggler #130: Add command line option to autostart publishers with all topics

bysteven, 2019-01-25:
I am setting up an automated system for rosbag analysis. I am launching plotjuggler with -l and -d, afterwards go to the menu to start the publishers. It would be really awesome if there was an option to autostart the publishers (all topics, or with a python statement like for rosbag filter).

For example:
rosrun plotjuggler PlotJuggler -l some_config -d some_file -p all

bysteven, 2019-01-25:
That would be evenly awesome.

facontidavide, 2019-01-25:
ok, I will add it to my TODo list

bysteven, 2019-01-27:
Great, thank you!

bysteven, 2019-01-27:
How can I tip you with a beer? Do you have a XRP tipbot address or similar?

facontidavide, 2019-01-27:
My PayPal account can be found in the Readme file. Cheers!

facontidavide, 2019-05-29:
feature has been implemented in branch "parsers" that will be merged soon

closed 2019-06-21 by facontidavide