PlotJuggler/PlotJuggler #133: Save layout / open existing layout leads to error

bysteven, 2019-01-29:
Saving and opening of a layout resulted in the following error: Failed to parse snippets (xml), error unexpected end of file at line 1.

However, functionality is still given, the correct layout is saved and also restored.

facontidavide, 2019-01-29:
Can you please share the layout file here to reproduce the error?

bysteven, 2019-01-29:
Hi Davide,

I have attached the layout file. This also happens when opening it with the -t option. Thanks for having a look!

[plotjuggler_tmux_rosbag_analysis.xml.gz](https://github.com/facontidavide/PlotJuggler/files/2807787/plotjuggler_tmux_rosbag_analysis.xml.gz)

bysteven, 2019-01-29:
I have pulled the release from yesterday and it does not occur anymore.

bysteven, 2019-01-29:
Thanks anyway! :)

closed 2019-01-29 by bysteven