PlotJuggler/PlotJuggler #134: Topic List not showing subscribed topics.

d-walsh, 2019-02-06:
After subscribing to topics they only appear in the topic list after a message has been received. However I would like to create my plots before the data has been received. It's very confusing as a user to not see the topics in the list. Is this the intended functionality? You could have a checkbox for showing only topics that have received data?

facontidavide, 2019-02-06:
I should probably put this in a FAQ...

PlotJuggler is able to read any existing ROS message, even user defined one, without any previous knowledge at compilation time. The price to pay is that the fields in any message can be known only once at least a single message has been received.

This problem COULD be partially addressed making some changes for messages which dimension is statis (Pose, Quaternion, Header) but in the __vast__ mayority of the cases the messages contain vectors which size is unknown before the message is received.

This is neither "a feature" not "a bug", it is an intrinsic limitation.

For more information: https://github.com/facontidavide/ros_type_introspection

closed 2019-02-06 by facontidavide

d-walsh, 2019-02-10:
Thanks for explaining. I think this places the problem as a UI issue then - as users can't easily understand what PlotJuggler is doing. A potential fix would be to show the subscribed topics that haven't received any messages in the list but greyed out / disabled. What do you think?

facontidavide, 2019-02-11:
That would require more changes in the code than you would expect.

I have a new feature planned for the future that use the topics name to create a more compact tree view, (related to #74). That would solve these two problems at once.