PlotJuggler/PlotJuggler #135: Scaling x-axis independently from y

skohlbr, 2019-02-06:
When loading long bagfiles and maximizing the y-axis vertical zoom, datapoints can get very crowded on the x-axis. It would be very useful to keep the y-axis scaling as is, but be able to "stretch" the x-axis via the mouse. This is a feature that's available in [rqt_multiplot](https://github.com/anybotics/rqt_multiplot_plugin) by holding the right mouse-button and moving the mouse left/right and it would be awesome to have it in PlotJuggler.

One can achieve a similar effect by manually specifying (high) y axis limits that "squish" the y axis down, but this is fairly cumbersome.

I searched for similar issues and talked to other PlotJuggler users about below feature request, apologies if I missed existing discussions, or even the existence of the feature ;)

facontidavide, 2019-02-06:
No need to apologies, the feature does not exist...

The way I would do it is through area zoom, i.e. selecting the rectangle to zoom, but I juess you are refering to the zoom with the Mouse Wheel, right?

It make sense. I guess I will implement it at some point. Which combination of  KEY + MOUSE do you suggest? Does **SHIFT + Mouse Wheel** work for you?

facontidavide, 2019-02-06:
ok.

pressing the **key X + Wheel mouse** will scale only the X axis.
pressing the **key Y + Wheel mouse** will scale only the Y axis.

facontidavide, 2019-02-06:
About "talked to other PlotJuggler users"... Where did you find them?
😆

facontidavide, 2019-02-06:
it is still a little bit buggy :( I have to fix a couple of things

facontidavide, 2019-02-06:
ok, NOW it works

closed 2019-02-06 by facontidavide

skohlbr, 2019-02-06:
> About "talked to other PlotJuggler users"... Where did you find them?

In our lab :) Lately, it seems that PlotJuggler overtakes other plotting tools in popularity due the ease of use, so I asked around if I just missed how to scale axes separately.

Just compiled from source and tried it. Works like a charm, thanks a lot! It would probably be a good idea to document the feature somewhere, as it otherwise would be hard to discover.

facontidavide, 2019-02-06:
yes, I am preparing a cheatsheet... https://www.plotjuggler.io/cheatsheet

facontidavide, 2019-02-07:
> In our lab :) Lately, it seems that PlotJuggler overtakes other plotting tools in popularity due the ease of use

If someone in your lab wants to do a proper "testimonial" that would be nice, since I am plaaning to include it in the web page (I want to raise money to develop PlotJuggler even further).
You can find my email in my Github profile.

facontidavide, 2019-03-25:
For the records. I decided to  change the behavior of the single axis zoom. Please check the cheatsheet or this video https://github.com/facontidavide/PlotJuggler/blob/master/plotter_gui/cheatsheet/video/cheatsheet-zoom-in-out.mp4

xdaumich, 2022-06-01:
@facontidavide I am trying to use single axis zoom. I have try "pressing the key X + Wheel mouse will scale only the X axis.
pressing the key Y + Wheel mouse will scale only the Y axis." It does not work. The video link is not working. https://github.com/facontidavide/PlotJuggler/blob/master/plotter_gui/cheatsheet/video/cheatsheet-zoom-in-out.mp4

I just compile from the source. Thanks for the help.

MattMgn, 2024-03-28:
same here, i cannot zoom on a single axis, the version i use is 3.7.1, any idea of what is going on? (i could open an issue if needed)