PlotJuggler/PlotJuggler #139: Doesn't build on ROS Melodic

maxlein, 2019-02-11:
I have an "multiple definition" error on RosIntrospection:

`  
CMakeFiles/RosTopicPublisher.dir/RosTopicPublisher_autogen/mocs_compilation.cpp.o: In function RosIntrospection::ShapeShifter::size() const:  
/usr/include/boost/flyweight/static_holder.hpp:33: multiple definition of RosIntrospection::ShapeShifter::size() const  
CMakeFiles/RosTopicPublisher.dir/TopicPublisherROS/statepublisher_rostopic.cpp.o:/opt/ros/melodic/include/ros_type_introspection/utils/shape_shifter.hpp:224: first defined here
collect2: error: ld returned 1 exit status  
PlotJuggler/plugins/ROS/CMakeFiles/RosTopicPublisher.dir/build.make:181: recipe for target '/home/kin/ws_plotjuggler/devel/lib/plotjuggler/libRosTopicPublisher.so failed
make[2]: *** [/home/kin/ws_plotjuggler/devel/lib/plotjuggler/libRosTopicPublisher.so] Error 1
CMakeFiles/Makefile2:2576: recipe for target PlotJuggler/plugins/ROS/CMakeFiles/RosTopicPublisher.dir/all failed  
make[1]: *** [PlotJuggler/plugins/ROS/CMakeFiles/RosTopicPublisher.dir/all] Error 2  
make[1]: *** Waiting for unfinished jobs..    
`
Using Ros melodic, building from source like described in README.  
Also using apt install for plotjuggler installs something, but no binary to execute ...

facontidavide, 2019-02-11:
The latest plotjuggler needs the version 1.3.0 of ros_type_introspection.

As you can see here, it will be released soon.

http://repositories.ros.org/status_page/ros_melodic_default.html?q=SYNC

closed 2019-02-11 by facontidavide

facontidavide, 2019-02-11:
https://discourse.ros.org/t/new-packages-for-melodic-2019-02-11/7809

Update ros_type_introspection and it should work....