PlotJuggler/PlotJuggler #140: [TODO]: Tree view (issue #134 and #74]

facontidavide, 2019-02-11:
Add a way to group timeseries into groups and show them in a tree view with a single depth level.

for instance, give the topic __/turtle1/pose__

istead of :

>       /turtle1/pose/linear_velocity
>       /turtle1/pose/theta
>       /turtle1/pose/ x
>       /turtle1/pose/y

Weshould see a tree with this structure

>     /turtle1/pose
>     -- linear_velocity
>     -- theta
>     -- x
>     -- y

bresch, 2019-02-28:
Related to that issue, in `.ulg` files, the signals that have more than a single depth level are not shown properly:

Currently in PlotJuggler:
[image]

and those triplets should contain:
[image]

facontidavide, 2019-02-28:
@bres The error is not related to this issue. I will try to to fix it though

facontidavide, 2019-11-10:
I am extremely excited to close this issue

closed 2019-11-10 by facontidavide