PlotJuggler/PlotJuggler #141: Does not run on Ubuntu 14.04 trusty

probonopd, 2019-02-15:
```
AppRun: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.20' not found (required by /run/firejail/appimage/.appimage-10555/AppRun)
/run/firejail/appimage/.appimage-10555/AppRun: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by /run/firejail/appimage/.appimage-10555/AppRun)
```

Please see https://github.com/AppImage/AppImageKit/wiki/Creating-AppImages#creating-appimages-that-are-compatible-with-many-systems

Reference:
https://github.com/AppImage/appimage.github.io/pull/1266

Full test log:
https://api.travis-ci.org/v3/job/492839113/log.txt

facontidavide, 2019-02-15:
That is somehow expected, the app image was built on Ubuntu 16.04
Indigo and trusty will be EOL soon, therefore I will not bother to keep compatibility with it any longer. You may find instructions to build your app image here https://github.com/facontidavide/PlotJuggler/blob/master/appimage_howto.md

facontidavide, 2019-02-15:
By the way, I know you don't need instructions to use your own software... I included the link for other people. Great work with linuxqtdeploy

probonopd, 2019-02-15:
Thanks for the clarification. Since being listed on appimage.github.io requires passing a test on the oldest still-supported Ubuntu release, we should revisit this once trusty has reached EOL.

closed 2019-02-18 by facontidavide