PlotJuggler/PlotJuggler #142: several icons missed

wt70707, 2019-02-17:
I use ubuntu 16.04, ROS Kinetic. I like the tool. Recently it seems upgraded for the automatically update. However, some of the icons disappeared including the "add a new row", now on the right (before is on the left) there is only 3 icons: clean, legend and link. How should I do?

facontidavide, 2019-02-17:
Oh snap!

It seems to be a problem with the SVG icons... but it is not supposed to happen! I guess your installed version is 2.0.4, can you confirm?

Please try to do

          sudo apt install libqt5svg5-dev

- Was it already installed in your system or not?
- Does it solve the problem?

wt70707, 2019-02-17:
I already installed that. I got "libqt5svg5-dev is already the newest version (5.5.1-2build1)" when install it again. 
It is version 2.0.4 as I installed with "sudo apt-get install ros-kinetic-plotjuggler". I also follow your instructions and clone the master branch which goes to version 2.1.2, but cannot see the missed 6 icons and now there is a vertical line after the left missed 3 icons (it can be reached if I put the mouse on there, this is why In know 6 icons missed).

facontidavide, 2019-02-17:
Ok, this is weird... 

You are experiencing a corner case that I can hardly debug. Do you have any special theme installed in Ubuntu?

Whatever the answer is, I should probably remove the SVG icons in version 2.1.3 ...

Can you please download and run the AppImage and tell me if the icons are visible? 

https://github.com/facontidavide/PlotJuggler/releases/download/2.1.2/PlotJuggler-2.1.2-x86_64.AppImage

wt70707, 2019-02-17:
I run the AppImage got:
"wt70707@wt70707xps:~/Downloads$ ./PlotJuggler-2.1.2-x86_64.AppImage 
./PlotJuggler-2.1.2-x86_64.AppImage: /opt/qt57/lib/libQt5Core.so.5: version `Qt_5.9' not found (required by ./PlotJuggler-2.1.2-x86_64.AppImage)"

Then I check qt which shows it is qt_5.5, should I upgrade to 5.9? Do you the quick way to do it?

wt70707@wt70707xps:~/Downloads$ sudo apt-get -y install qtbase5-dev libqt5svg5-dev qtdeclarative5-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libqt5svg5-dev is already the newest version (5.5.1-2build1).
qtdeclarative5-dev is already the newest version (5.5.1-2ubuntu6).
qtbase5-dev is already the newest version (5.5.1+dfsg-16ubuntu7.5).

facontidavide, 2019-02-18:
the AppImage is supposed to work out of the box, you are not supposed to install anything :(

I will try to address both these problems, thanks for the help.

facontidavide, 2019-02-18:
I have tried the AppImage on a freshly installed Ubuntu Xenial and it works :(

Furthermore, I can see all the icons. It must be something very specific of your system.

Nevertheless, I will remove all the SVG icons in the future, just in case.

closed 2019-02-18 by facontidavide

wt70707, 2019-02-18:
Thanks Davide. Yes I use ROS a lot and installed Ecllipse and Qt several times to get some packages work. I cannot discern what thing makes this problem. Please update me when you make the change.

facontidavide, 2019-02-18:
version 2.1.3 introduce the change (already released).

I am building the AppImage...

wt70707, 2019-02-18:
On branch master? I clone yesterday it is still 2.1.2.

facontidavide, 2019-02-18:
https://github.com/facontidavide/PlotJuggler/releases/download/2.1.3/PlotJuggler-2.1.3-x86_64.AppImage

https://github.com/facontidavide/PlotJuggler/releases/tag/2.1.3

wt70707, 2019-02-18:
I put 2.1.3 into my catkin workspace, now the icons come back. But the AppImage is still the same:
wt@wt:~/Downloads$ ./PlotJuggler-2.1.3-x86_64.AppImage 
./PlotJuggler-2.1.3-x86_64.AppImage: /opt/qt57/lib/libQt5Core.so.5: version `Qt_5.9' not found (required by ./PlotJuggler-2.1.3-x86_64.AppImage)

facontidavide, 2019-02-18:
Sorry, I don't know how to help you

wt70707, 2019-02-18:
You already did. I work most of time in Ros, and 2.1.3 already works.
AppImage is just for test., and maybe it can help me find problem of my
system. Thanks!