PlotJuggler/PlotJuggler #145: Custom calculation on a msg struct (rather than a msg endpoint) as source data

koenlek, 2019-02-20:
First of all, thanks for the great 'custom timeseries' feature! It is really cool and has proven to be very useful for me already! 

If I understand correctly, the 'source data' can currently only give you access to 1 datapoint, existing of a timestamp and a value. The source data seems to always be a and 'endpoint' of a topic, i.e. the value will never point to a struct, always just a basic datatype (int, float/double, string). 

That makes me wonder how the example snippet `yaw_from_quaternion` is supposed to be used. It needs the values x,y,z,w, which are parsed with the double dollars (`$$x$$`). I believe you call such variables 'channels' or 'curves'. However, how can I get all values from 1 quaternion in these 4 channel variables? The only way I know would be to replace them each. E.g. for a topic `/foo` with `pose` filed of type `geometry_msgs/Pose` I'd change:

* `$$x$$` to `$$/foo/pose/orientation/x$$`
* `$$y$$` to `$$/foo/pose/orientation/y$$`
* `$$z$$` to `$$/foo/pose/orientation/z$$`
* `$$w$$` to `$$/foo/pose/orientation/w$$`

That's a bit clumsy I think. I'm looking forward to build a set of useful functions based on certain message types. Hence, it would be nicer if I could don one of these things:

1. Select a struct (parent, rather than endpoint) as source data.
2. Select a struct as channel variable.
3. Get creative with eval to separate prefix (topic name + fields) from variable names.

I tried all 3 options and none of them work yet. 

**Option 1: select struct from 'source data' list**
A suggestion could be to select from a tree view, similar to your idea in #140. For the tree view, instead of a flat dropdown list you could consider either adding fold/unfold buttons, or expanding menus (like the top menu File -> Open Recent -> (list of recents)).
Such a tree view would be useful to add to the timeseries list of the main window and the source data and curves lists of the 'create custom timeseries' window. The important part for me would be the ability to select a struct, rather than an endpoint.

**Option 2: Select a struct as channel variable.**

I tried this. What I did in the function calc body:

```js
var quat = $$/foo/pose/orientation$$
// and then do e.g. 
return quat.x
// or something more complex ;)
```

Whenever I just try setting something like `var quat = $$/foo/pose/orientation$$` I get the error, 'invalid channel name' as soon as I hit 'Create New Timeseries'.

**Option 3: Get creative with eval**

This is not a serious option, but rather a quick workaround I tried. Surprisingly enough, I couldn't get it to work:

```js
var x1 = $$/foo/pose/orientation/x$$ // works!
var x2 = eval('$$/foo/pose/orientation/x$$') // works!

var channel_string = '$$/foo/pose/orientation/x$$'
var x3 = eval(channel_string) // works!

var x4 = eval('$$/foo/pose/orientation/'+'x$$') // doesn't work (invalid channel name)... I don't understand why
var channel_string_left = '$$/foo/pose/orientation/'
var channel_string_right = 'x$$'
var channel_string_total = channel_string_left.append(channel_string_right)
var x5 = eval(channel_string_total) // doesn't work (invalid channel name)... I don't understand why

// and what I actually wanted:
var prefix = '/foo/pose/orientation/'
var x5 = eval('$$'+prefix+'x'+'$$') // also doesn't work (invalid channel name)... I don't understand why
```

**Finally**

It would be great if you could make option 1 or 2 to work. I saw that rosmsg generates js libraries for all message types too, so having the structs available in js shouldn't be an issue.

Of course I understand this is still experimental functionality. I'll just patiently wait until this feature request is implemented (if ever)! Just wanted to point it out, as I guess more users would enjoy such functionality. 

And again: thanks a lot for the custom timeseries functionality so far: it's really useful! 

**Setup data**

Used latest version of master (a8743184bd2b01f2862f9ca46bfd5e81e8a035e5)
ROS Kinetic in Ubuntu 16.04.

facontidavide, 2019-02-20:
Hi,

first of all, Thank you very much for your suggestion and ideas.

I know that we might do better, but it is not trivial.

Please remember that PlotJuggler is NOT a ROS application and it will never be.
ROS is supported by plugins.

PlotJuggler is not aware of what a message is, in other words, it doesn't uderstand that `/foo/pose/orientation/x` is somehow related to `/foo/pose/orientation/y`.

PJ does not make any assumption (for the time being) about the relationship between individual timeseries; that is by design.

As you mentioned, this may change with #140 and ironically it might move the code base one step closer to improvinf this issue as well.

Still, option 2) seems to me quite complicated to implement.

At some point I might implement option 1)...

But it will take time, since right not 80% of my time goes to find a way to support economically PlotJuggler.

koenlek, 2019-02-23:
Okay, I didn't realize the ros-less aspect. Makes sense.

I'll just patiently wait and hope that you'll find the time to implement #140 and then build this request on top of that at some point :)

facontidavide, 2019-06-21:
I don't expect this issue / suggestion for improvement to be implemented anytime soon. I am closing this for the time being.

closed 2019-06-21 by facontidavide