PlotJuggler/PlotJuggler #146: [Feature request] Support multiple instances of the same message

bresch, 2019-02-21:
Hi, I'm trying to move from FlightPlot to PlotJuggler to read `.ulg` files as I see a huge potential in that project.

The issue is the following:
Some messages have multiple instances and they are plotted as a single signal:
[image]

The following result is expected:
[image]

For completeness, this is the log file:
https://logs.px4.io/plot_app?log=412f7d6b-3d3f-42e9-b04f-c9a55e16bf9d

Many thanks!

facontidavide, 2019-02-21:
No problem, I can fix this. Can you please send me the file to be able to reproduce this?

facontidavide, 2019-02-21:
No problem, I should be able to fix this easily. Thanks for reporting

facontidavide, 2019-02-21:
Fixed it:

[image]

Can you please check out the last commit and test it?

facontidavide, 2019-02-21:
I am closing it. Let me know if you have any additional issue.

i will soon release 2.1.4 with this improvement

closed 2019-02-21 by facontidavide

bresch, 2019-02-22:
@facontidavide Awesome! That was quick!