PlotJuggler/PlotJuggler #147: ulog analysis shows alternating initial values on some topics

MaEtUgR, 2019-02-21:
> - Please use always the master branch or the latest tag, if you can.

Sorry but I used `PlotJuggler-2.1.3-x86_64.AppImage` for my first testing. Feel free to close this issue when you have any suspicion that problem could already be fixed on master.

> - Check if the problem can be reproduced using the dummy data created by the command line argument "-t" or one of the files in the folder "datasamples".

I tested with https://github.com/facontidavide/PlotJuggler/blob/master/datasamples/sample.ulg.zip and didn't see the prolem with any topic in there.

> - If it can't be reproduced with the dummy data, __please__ share the CSV file or the rosbag that can be used to reproduce the problem.

Sure, it can for example be reproduced by loading the 

**Steps to reproduce**
1. Download the [latest uploaded log I found from the PX4 teast team](https://logs.px4.io/download?log=51ba2617-5abd-444d-b665-aa964cde1546)
2. Open it in PlotJuggler with `File` > `Load Data`
3. From the topic `battery_status/connected`
4. Even though this field should always show the same value of 1 because the battery was connected the whole flight you see alternating 0 and 1 in the plot:
    [image]
    zoomed in:
    [image]

**Verification**
I used [FlightPlot](https://docs.px4.io/en/log/flight_log_analysis.html#flightplot-desktop) to check that such alterning values are not really present in this exact log:
[image]
[image]

**Additional Findings**
- It likely has something to do with the initial value of the field because e.g. `battery_status/remaining` starts with 1 (full battery), goes down and the alternating value is 1.
[image]
- I saw this same problems in other logs with different topics so it can't be exclusive to the `battery_status`.
- Just a far guess: It might have something to do with time series in ulogs having flexible sampeling. Some topics update faster some slower and hence updates can in theory happen at arbitrary timestamps. Then it would likely happen with slower updated topics but I didn't verify this. (@stifael was recently looking into this whole topic for automatic log analysis in python: https://github.com/Stifael/uloganalysis/blob/master/uloganalysis/dfUlg.py#L43-L57)

MaEtUgR, 2019-02-21:
Now that I saw it this could be the same issue as https://github.com/facontidavide/PlotJuggler/issues/146 I just didn't spot the cause.

facontidavide, 2019-02-21:
fixed 30 seconds ago :)

facontidavide, 2019-02-21:
It is a duplicate of #146

closed 2019-02-21 by facontidavide

facontidavide, 2019-02-21:
By the way, let me thank you for the very detailed error report. I really appreciate the way you spent time to make the issue clear and repeatable

MaEtUgR, 2019-02-24:
Thanks! You're very fast 👍