PlotJuggler/PlotJuggler #148: Mac OS install compatibility?

ryanjAA, 2019-02-23:
Was trying to install on Mac OS (read the comments from August 2018) but couldn't get it to build, QT installed for PX4 development work but am excited to try this. 

1 - Was the issue fixed listen in the comments from 2018?

2 - Any install notes/guide for Mac implementation?

Looking forward to trying it!

facontidavide, 2019-02-23:
Sorry, I have zero experience in terms of development on Mac OS, therefore I am 90% sure that compiling on Mac should be easy (all my code is supposed to be portable), but I have no means to test it. 

Can you give me more details about the error?

facontidavide, 2019-03-01:
@ryanjAA I am not going to work on this, in particular if you don't give me any detail about the compilation error. 

If you manage to fix compilation on Mac OS yourself, please send a Pull Request.

zhaoxiaowei1013, 2019-09-06:
it is can use normally in mojove 10.14.6 with dark mode .

ryanjAA, 2019-09-06:
What are you doing to get this to work? Take a look at the output here:

make
Scanning dependencies of target colorwidgets_autogen
[  1%] Automatic MOC and UIC for target colorwidgets
[  1%] Built target colorwidgets_autogen
Scanning dependencies of target colorwidgets
[  2%] Building CXX object color_widgets/CMakeFiles/colorwidgets.dir/colorwidgets_autogen/mocs_compilation.cpp.o
[  3%] Building CXX object color_widgets/CMakeFiles/colorwidgets.dir/src/color_names.cpp.o
[  3%] Building CXX object color_widgets/CMakeFiles/colorwidgets.dir/src/color_palette.cpp.o
[  4%] Building CXX object color_widgets/CMakeFiles/colorwidgets.dir/src/color_utils.cpp.o
[  4%] Building CXX object color_widgets/CMakeFiles/colorwidgets.dir/src/color_preview.cpp.o
[  5%] Building CXX object color_widgets/CMakeFiles/colorwidgets.dir/src/color_wheel.cpp.o
[  6%] Building CXX object color_widgets/CMakeFiles/colorwidgets.dir/src/swatch.cpp.o
[  6%] Linking CXX static library ../lib/libcolorwidgets.a
[  6%] Built target colorwidgets
Scanning dependencies of target qwt_static_autogen
[  7%] Automatic MOC and UIC for target qwt_static
[  7%] Built target qwt_static_autogen
Scanning dependencies of target qwt_static
[  7%] Building CXX object qwt/src/CMakeFiles/qwt_static.dir/qwt_static_autogen/mocs_compilation.cpp.o
[  8%] Building CXX object qwt/src/CMakeFiles/qwt_static.dir/qwt_curve_fitter.cpp.o
[  9%] Building CXX object qwt/src/CMakeFiles/qwt_static.dir/qwt_bezier.cpp.o
[  9%] Building CXX object qwt/src/CMakeFiles/qwt_static.dir/qwt_spline_curve_fitter.cpp.o
[ 10%] Building CXX object qwt/src/CMakeFiles/qwt_static.dir/qwt_weeding_curve_fitter.cpp.o
[ 10%] Building CXX object qwt/src/CMakeFiles/qwt_static.dir/qwt_abstract_legend.cpp.o
[ 11%] Building CXX object qwt/src/CMakeFiles/qwt_static.dir/qwt_legend.cpp.o
[ 12%] Building CXX object qwt/src/CMakeFiles/qwt_static.dir/qwt_legend_data.cpp.o
[ 12%] Building CXX object qwt/src/CMakeFiles/qwt_static.dir/qwt_legend_label.cpp.o
[ 13%] Building CXX object qwt/src/CMakeFiles/qwt_static.dir/qwt_plot.cpp.o
[ 13%] Building CXX object qwt/src/CMakeFiles/qwt_static.dir/qwt_plot_renderer.cpp.o
[ 14%] Building CXX object qwt/src/CMakeFiles/qwt_static.dir/qwt_plot_xml.cpp.o
[ 15%] Building CXX object qwt/src/CMakeFiles/qwt_static.dir/qwt_plot_axis.cpp.o
[ 15%] Building CXX object qwt/src/CMakeFiles/qwt_static.dir/qwt_plot_curve.cpp.o
[ 16%] Building CXX object qwt/src/CMakeFiles/qwt_static.dir/qwt_plot_dict.cpp.o
/Users/ryanjohnston2/src/plotjugnew/PlotJuggler/qwt/src/qwt_plot_dict.cpp:25:17: warning: 'qUpperBound<QList<QwtPlotItem *>::iterator, QwtPlotItem *,
      QwtPlotDict::PrivateData::ItemList::LessZThan>' is deprecated: Use std::upper_bound [-Wdeprecated-declarations]
                qUpperBound( begin(), end(), item, LessZThan() );
                ^
/usr/local/opt/qt5/lib/QtCore.framework/Headers/qalgorithms.h:280:1: note: 'qUpperBound<QList<QwtPlotItem *>::iterator, QwtPlotItem *,
      QwtPlotDict::PrivateData::ItemList::LessZThan>' has been explicitly marked deprecated here
QT_DEPRECATED_X("Use std::upper_bound") Q_OUTOFLINE_TEMPLATE RandomAccessIterator qUpperBound(RandomAccessIterator begin, RandomAccessIterator end...
^
/usr/local/opt/qt5/lib/QtCore.framework/Headers/qglobal.h:294:33: note: expanded from macro 'QT_DEPRECATED_X'
#  define QT_DEPRECATED_X(text) Q_DECL_DEPRECATED_X(text)
                                ^
/usr/local/opt/qt5/lib/QtCore.framework/Headers/qcompilerdetection.h:649:55: note: expanded from macro 'Q_DECL_DEPRECATED_X'
#    define Q_DECL_DEPRECATED_X(text) __attribute__ ((__deprecated__(text)))
                                                      ^
/Users/ryanjohnston2/src/plotjugnew/PlotJuggler/qwt/src/qwt_plot_dict.cpp:35:17: warning: 'qLowerBound<QList<QwtPlotItem *>::iterator, QwtPlotItem *,
      QwtPlotDict::PrivateData::ItemList::LessZThan>' is deprecated: Use std::lower_bound [-Wdeprecated-declarations]
                qLowerBound( begin(), end(), item, LessZThan() );
                ^
/usr/local/opt/qt5/lib/QtCore.framework/Headers/qalgorithms.h:247:1: note: 'qLowerBound<QList<QwtPlotItem *>::iterator, QwtPlotItem *,
      QwtPlotDict::PrivateData::ItemList::LessZThan>' has been explicitly marked deprecated here
QT_DEPRECATED_X("Use std::lower_bound") Q_OUTOFLINE_TEMPLATE RandomAccessIterator qLowerBound(RandomAccessIterator begin, RandomAccessIterator end...
^
/usr/local/opt/qt5/lib/QtCore.framework/Headers/qglobal.h:294:33: note: expanded from macro 'QT_DEPRECATED_X'
#  define QT_DEPRECATED_X(text) Q_DECL_DEPRECATED_X(text)
                                ^
/usr/local/opt/qt5/lib/QtCore.framework/Headers/qcompilerdetection.h:649:55: note: expanded from macro 'Q_DECL_DEPRECATED_X'
#    define Q_DECL_DEPRECATED_X(text) __attribute__ ((__deprecated__(text)))
                                                      ^
2 warnings generated.
[ 16%] Building CXX object qwt/src/CMakeFiles/qwt_static.dir/qwt_plot_directpainter.cpp.o
[ 17%] Building CXX object qwt/src/CMakeFiles/qwt_static.dir/qwt_plot_grid.cpp.o
[ 18%] Building CXX object qwt/src/CMakeFiles/qwt_static.dir/qwt_plot_histogram.cpp.o
[ 18%] Building CXX object qwt/src/CMakeFiles/qwt_static.dir/qwt_plot_item.cpp.o
[ 19%] Building CXX object qwt/src/CMakeFiles/qwt_static.dir/qwt_plot_abstract_barchart.cpp.o
[ 19%] Building CXX object qwt/src/CMakeFiles/qwt_static.dir/qwt_plot_barchart.cpp.o
[... 1005 lines cut ...]
  "QwtPicker::trackerRect(QFont const&) const", referenced from:
[... cut ...]
      typeinfo for PlotMagnifier in mocs_compilation.cpp.o
  "typeinfo for QwtPlotLegendItem", referenced from:
      typeinfo for PlotLegend in mocs_compilation.cpp.o
  "typeinfo for QwtPlot", referenced from:
      typeinfo for PlotWidget in mocs_compilation.cpp.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [bin/PlotJuggler] Error 1
make[1]: *** [plotter_gui/CMakeFiles/PlotJuggler.dir/all] Error 2
make: *** [all] Error 2

zhaoxiaowei1013, 2019-09-06:
it is my output 
build git:(master) cmake ..
```
-- The C compiler identification is AppleClang 10.0.1.10010046
-- The CXX compiler identification is AppleClang 10.0.1.10010046
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- No build type selected, default to RelWithDebInfo
-- Could NOT find Bfd library named bfd.
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/xiaowei/src/PlotJuggler/build
```
➜  build git:(master) make
```
Scanning dependencies of target colorwidgets_autogen
[  0%] Automatic MOC and UIC for target colorwidgets
[  0%] Built target colorwidgets_autogen
Scanning dependencies of target colorwidgets
[  1%] Building CXX object color_widgets/CMakeFiles/colorwidgets.dir/colorwidgets_autogen/mocs_compilation.cpp.o
[  1%] Building CXX object color_widgets/CMakeFiles/colorwidgets.dir/src/color_names.cpp.o
[  2%] Building CXX object color_widgets/CMakeFiles/colorwidgets.dir/src/color_palette.cpp.o
[  2%] Building CXX object color_widgets/CMakeFiles/colorwidgets.dir/src/color_utils.cpp.o
[  3%] Building CXX object color_widgets/CMakeFiles/colorwidgets.dir/src/color_preview.cpp.o
[  4%] Building CXX object color_widgets/CMakeFiles/colorwidgets.dir/src/color_wheel.cpp.o
[  4%] Building CXX object color_widgets/CMakeFiles/colorwidgets.dir/src/swatch.cpp.o
[  5%] Linking CXX static library ../lib/libcolorwidgets.a
[  5%] Built target colorwidgets
Scanning dependencies of target qwt_static_autogen
[  6%] Automatic MOC and UIC for target qwt_static
[  6%] Built target qwt_static_autogen
Scanning dependencies of target qwt_static
[  7%] Building CXX object qwt/src/CMakeFiles/qwt_static.dir/qwt_static_autogen/mocs_compilation.cpp.o
[  8%] Building CXX object qwt/src/CMakeFiles/qwt_static.dir/qwt_curve_fitter.cpp.o
[  8%] Building CXX object qwt/src/CMakeFiles/qwt_static.dir/qwt_bezier.cpp.o
[  9%] Building CXX object qwt/src/CMakeFiles/qwt_static.dir/qwt_spline_curve_fitter.cpp.o
[  9%] Building CXX object qwt/src/CMakeFiles/qwt_static.dir/qwt_weeding_curve_fitter.cpp.o
[ 10%] Building CXX object qwt/src/CMakeFiles/qwt_static.dir/qwt_abstract_legend.cpp.o
[ 11%] Building CXX object qwt/src/CMakeFiles/qwt_static.dir/qwt_legend.cpp.o
[ 11%] Building CXX object qwt/src/CMakeFiles/qwt_static.dir/qwt_legend_data.cpp.o
[ 12%] Building CXX object qwt/src/CMakeFiles/qwt_static.dir/qwt_legend_label.cpp.o
[ 13%] Building CXX object qwt/src/CMakeFiles/qwt_static.dir/qwt_plot.cpp.o
[ 13%] Building CXX object qwt/src/CMakeFiles/qwt_static.dir/qwt_plot_renderer.cpp.o
[ 14%] Building CXX object qwt/src/CMakeFiles/qwt_static.dir/qwt_plot_xml.cpp.o
[ 14%] Building CXX object qwt/src/CMakeFiles/qwt_static.dir/qwt_plot_axis.cpp.o
[ 15%] Building CXX object qwt/src/CMakeFiles/qwt_static.dir/qwt_plot_curve.cpp.o
[ 16%] Building CXX object qwt/src/CMakeFiles/qwt_static.dir/qwt_plot_dict.cpp.o
/Users/xiaowei/src/PlotJuggler/qwt/src/qwt_plot_dict.cpp:25:17: warning:
      'qUpperBound<QList<QwtPlotItem *>::iterator, QwtPlotItem *,
      QwtPlotDict::PrivateData::ItemList::LessZThan>' is deprecated: Use std::upper_bound
      [-Wdeprecated-declarations]
                qUpperBound( begin(), end(), item, LessZThan() );
                ^
/usr/local/opt/qt5/lib/QtCore.framework/Headers/qalgorithms.h:280:1: note:
      'qUpperBound<QList<QwtPlotItem *>::iterator, QwtPlotItem *,
      QwtPlotDict::PrivateData::ItemList::LessZThan>' has been explicitly marked
      deprecated here
QT_DEPRECATED_X("Use std::upper_bound") Q_OUTOFLINE_TEMPLATE RandomAccessIterator ...
^
/usr/local/opt/qt5/lib/QtCore.framework/Headers/qglobal.h:294:33: note: expanded from
[... 610 lines cut ...]
[ 94%] Building CXX object plugins/DataLoadULog/CMakeFiles/DataLoadULog.dir/ulog_parser.cpp.o
[ 94%] Building CXX object plugins/DataLoadULog/CMakeFiles/DataLoadULog.dir/ulog_parameters_dialog.cpp.o
[ 95%] Linking CXX shared library ../../bin/libDataLoadULog.dylib
[ 95%] Built target DataLoadULog
Scanning dependencies of target DataStreamSample_autogen
[ 96%] Automatic MOC and UIC for target DataStreamSample
[ 96%] Built target DataStreamSample_autogen
Scanning dependencies of target DataStreamSample
[ 97%] Building CXX object plugins/DataStreamSample/CMakeFiles/DataStreamSample.dir/DataStreamSample_autogen/mocs_compilation.cpp.o
[ 98%] Building CXX object plugins/DataStreamSample/CMakeFiles/DataStreamSample.dir/datastream_sample.cpp.o
[ 98%] Linking CXX shared library ../../bin/libDataStreamSample.dylib
[ 98%] Built target DataStreamSample
Scanning dependencies of target DataStreamWebSocket_autogen
[ 98%] Automatic MOC and UIC for target DataStreamWebSocket
[ 98%] Built target DataStreamWebSocket_autogen
Scanning dependencies of target DataStreamWebSocket
[ 98%] Building CXX object plugins/DataStreamWebSocket/CMakeFiles/DataStreamWebSocket.dir/DataStreamWebSocket_autogen/mocs_compilation.cpp.o
[ 99%] Building CXX object plugins/DataStreamWebSocket/CMakeFiles/DataStreamWebSocket.dir/datastreamserver.cpp.o
[100%] Linking CXX shared library ../../bin/libDataStreamWebSocket.dylib
[100%] Built target DataStreamWebSocket
```
➜  build git:(master)

i think you maybe need to try solve linker error.
https://github.com/tpoechtrager/osxcross/issues/11

ryanjAA, 2019-09-06:
thanks - I will take a look now. My C compiler is GCC not CC so I will take a look at getting the linker error first sorted.

cmake ..
-- The C compiler identification is AppleClang 10.0.0.10001145
-- The CXX compiler identification is AppleClang 10.0.0.10001145
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/usr/bin/gcc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/usr/bin/gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/usr/bin/g++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/usr/bin/g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- No build type selected, default to RelWithDebInfo
-- Found BFD: /usr/local/lib/libbfd.a
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/ryanjohnston2/src/plotjugnew/PlotJuggler/build

zhaoxiaowei1013, 2019-09-07:
may be you need to install ```Command Line Tools```

closed 2019-10-09 by facontidavide

ryanjAA, 2019-10-26:
It was an issue on the 10.13 build environment. Building on 10.14.x works fine. Nothing special needs to be done and confirms works fine.

zhaoxiaowei1013, 2019-11-04:
10.15 catalina build ok。