PlotJuggler/PlotJuggler #150: [Feature request] Show PX4 parameters

bresch, 2019-02-28:
Hi, I'm trying to open an issue every time I have to go back to FlightPlot to find something I miss in PlotJuggler. Here is a new one:

I usually need to check the parameters of a `.ulg` file in order to understand the behavior of the drone. This information is available in FlightPlot and is really important to have:
[image]
It doesn't need to be nice, even a raw list is usually enough. A quick-search might be nice to have but I would prefer to have the list quickly without it if it is too much effort to have the quick-search.

Thanks a lot, that tool is awesome!

facontidavide, 2019-02-28:
Quick question. Are parameter fixed or are they changed dynamically?

facontidavide, 2019-03-01:
I just pushed this functionality 5e3f49ac6576cf5ee4799f5463efad9f78bab203

Take a look

facontidavide, 2019-03-01:
The know limitation is that it will deisplayed only static parameters and ignore dynamic ones.
You should live with that ;)

closed 2019-03-01 by facontidavide

bresch, 2019-03-01:
@facontidavide Awesome! I just tested it and it is perfect. One quick question: Can I reopen it if I close the window?

> Quick question. Are parameter fixed or are they changed dynamically?

Yes, but in most cases, no.

> The know limitation is that it will deisplayed only static parameters and ignore dynamic ones.
You should live with that ;)

Perfectly fine for now :)

facontidavide, 2019-03-01:
> Can I reopen it if I close the window?

I finished implementing this at 1:00 in the morning soooo.... no.

Seriously though, I implemented this in a **quick and dirty** way and I am going to shift my attention to other features of PJ, in particular those feature sponsored by  companies / paying clients.

If you want to improve this yourself let's talk about it in terms of UI design and feel free to send a Pull Request ;)

bresch, 2019-03-01:
@facontidavide I'm already amazed that you did that over night :D