PlotJuggler/PlotJuggler #151: Error parsing position_setpoint_triplet

facontidavide, 2019-02-28:
@bresch

[image]

facontidavide, 2019-02-28:
that was though....

closed 2019-02-28 by facontidavide

bresch, 2019-03-01:
@facontidavide Works perfectly, thanks!

facontidavide, 2019-03-01:
You are welcome. I must confess I skimmed through the documentation of the Ulog format and I did not realized that a Format can contain other Format in a nested way.

Solved it with classical recursion ;)

junwoo091400, 2022-10-18:
> You are welcome. I must confess I skimmed through the documentation of the Ulog format and I did not realized that a Format can contain other Format in a nested way.

FYI the ULog Format doc got a recent upgrade, in case you are interested 😉 @facontidavide 
https://docs.px4.io/main/en/dev_log/ulog_file_format.html