PlotJuggler/PlotJuggler #153: Inverted mouse wheel zoom

bkueng, 2019-03-04:
Hi

Really nice tool! Thanks for your efforts.
I have some usability points I'd like to discuss:
- The mouse scroll zoom is inverted (at least on Linux). Can you invert it or add an option to do so?
- There's already a button for zooming out completely. However adding some mouse event (like double-click on the plot) would make the workflow more efficient.
- The individual axis zoom via x/y keys is nice. I suggest extending it: zoom the corresponding axis when hovering over an axis and scroll. Currently this does nothing, so there would be no conflict.

facontidavide, 2019-03-04:
Thank you for your suggestions :)

- Yes, you are right. I might probably invert it
- I am not planning to add this, for the time being. But I might change my mind at any point ;)
- Nice idea! I will do that.

Summarizing, I will work on these enhancement, but not in the short term.

RomanBapst, 2019-03-14:
This would be highly appreciated! Would facility my daily work immensely.

facontidavide, 2019-03-19:
done

closed 2019-03-19 by facontidavide