PlotJuggler/PlotJuggler #156: Topic with a float32 NaN lead to and crash

FabianUhl, 2019-03-08:
Hi, 
I m currently working at project with a sensor Battery state msg.  If I  preload a float32 with a NaN and try to plot this msg with plotjuggler, it will run into an issue. This Issue can be reproduced with

 `rostopic pub -r 1 /test std_msgs/Float32 '.NAN'` 

Error Report:
> Failed to launch the streamer ....
terminate called after throwing an instance of 'RosIntrospection::RangeException'
what():  Floating point truncated

It is exclusively an issue with float32. The same with float64 works fine.

Best regards

facontidavide, 2019-03-08:
I will have a look at this today. Thanks for reporting

facontidavide, 2019-03-08:
Fixed. NAN and INF should be ignored now

closed 2019-03-08 by facontidavide