PlotJuggler/PlotJuggler #158: PlotJuggler gets slow with time

RomanBapst, 2019-03-14:
I noticed that when I have PlotJoggler open for a while it gets slower as time passes. When e.g. trying to open a new log file it takes a while for the window to open...
It feels like the running a program on a device where the CPU is busy looping.
Happy to provide more information and to look into it.

facontidavide, 2019-03-15:
Interesting, I will look at it

facontidavide, 2019-03-19:
I am back from my holidays and I would like to address this issue. Can you describe a way to reproduce it, maybe sharing with me your log file?

facontidavide, 2019-03-27:
Hi,

unless you provide some more information about how to reproduce this, I will probably close this issue for the time being.

closed 2019-04-17 by facontidavide