PlotJuggler/PlotJuggler #160: Comparing data in stacked plots can be difficult due to misalignment

skohlbr, 2019-03-21:
When plotting data with different magnitudes on their y  scales into two plots in separate rows stacked above each other, the fact that the y axis legend is wider for one of them means that the grid on the plots is slightly offset between them, which makes comparing values a lot more difficult than it could be.

In principle, the vertical time tracker could be used, but with very long bag files it cannot be controlled accurately enough with the slider at the bottom to be useful. So either aligning plots, or being able to move the vertical tracker using the mouse would make this comparison use case easier.

See plot below (and the hand-drawn red marking on the left middle showing the offset)
[image]

This is using PlotJuggler from source/master.

facontidavide, 2019-03-21:
Good point. I should take care of this at some point

aeudes, 2019-03-21:
for completeness, please also note that the tracker could already be move with shift+click

skohlbr, 2019-03-21:
> for completeness, please also note that the tracker could already be move with shift+click

Oh, I missed that, excellent :)

facontidavide, 2019-03-21:
@aeudes thanks for pointing that out. As documented in the Cheatsheet, the most convenient way to move the tracker is Shift + Left Mouse

facontidavide, 2019-10-09:
I will not work on this issue, probably. Closing it for the time being

closed 2019-10-09 by facontidavide