PlotJuggler/PlotJuggler #166: Unable to open ROS bags, can only select .ulg files

ndepal, 2019-03-25:
I just started using PlotJuggler to analyze PX4 .ulg files.

I now wanted to open a ROS bag. In the `File > Load Data` dialog, I can only select "Files of type" `*.csv, *.ulg` and therefore can't load a ROS bag file.

I'm sure I'm missing something obvious. What do I need to do?
Thanks!

This is with versions 2.1.5 and 2.1.8.

facontidavide, 2019-03-25:
Are you compiling it or using the AppImages?

ndepal, 2019-03-25:
AppImages

facontidavide, 2019-03-25:
take a look at the console message, this is the "correct one":

> "libDataLoadCSV.so" : is a DataLoader plugin
> "libDataLoadROS.so" : is a DataLoader plugin
> "libDataLoadULog.so" : is a DataLoader plugin
> "libDataStreamROS.so" : is a DataStreamer plugin
> "libDataStreamSample.so" : is a DataStreamer plugin
> "libDataStreamSample.so" ...but will be ignored unless the argument -t is used.
> "libRosoutPublisher.so" : is a StatePublisher plugin
> "libRosTopicPublisher.so" : is a StatePublisher plugin

ndepal, 2019-03-25:
```"libDataLoadCSV.so" : is a DataLoader plugin
"libDataLoadROS.so" :  "Cannot load library /tmp/.mount_PlotJuRqpCar/libDataLoadROS.so: (libros_type_introspection.so: cannot open shared object file: No such file or directory)"
"libDataLoadULog.so" : is a DataLoader plugin
"libDataStreamROS.so" :  "Cannot load library /tmp/.mount_PlotJuRqpCar/libDataStreamROS.so: (libros_type_introspection.so: cannot open shared object file: No such file or directory)"
"libDataStreamSample.so" : is a DataStreamer plugin
"libDataStreamSample.so" ...but will be ignored unless the argument -t is used.
"libRosoutPublisher.so" : is a StatePublisher plugin
"libRosTopicPublisher.so" :  "Cannot load library /tmp/.mount_PlotJuRqpCar/libRosTopicPublisher.so: (libros_type_introspection.so: cannot open shared object file: No such file or directory)"
```

facontidavide, 2019-03-25:
it is what I expected :(

Do:
      sudo apt install ros-kinetic-ros-type-introspection 

Changing __kinetic__ with your ROS distro if necessary. I haven't been able to fix this in the AppImage yet.

ndepal, 2019-03-25:
That fixed it. Thanks for the fast reponse!

ndepal, 2019-03-25:
Well actually, it works if I run PlotJuggler form a terminal. If I use a `.desktop` file to run it, it does not work...

facontidavide, 2019-03-25:
The right solution would be to embed the ros_type_introspection library into the AppImage, but I haven't managed to do it for some reason.... any help is welcomed.

These are the steps I follow to create the AppImage: https://github.com/facontidavide/PlotJuggler/blob/master/appimage_howto.md

facontidavide, 2019-10-04:
Fixed!!!!!!!!!!!

closed 2019-10-04 by facontidavide

facontidavide, 2019-10-16:
I completely lied to you. App Image 2.3.6 is the first one that really fix this issue