PlotJuggler/PlotJuggler #168: Segmentation fault when creating custom timeseries that use data from multiple messages in ulg

ndepal, 2019-03-29:
I am trying to use the custom timeseries feature with a `ulg` file.
I want to write a function whose output depends on fields from multiple different messages.
Creating a custom timeseries that only uses the source data works fine.
When adding another curve from the curves tab and clicking "Create new time series" or "Modify timeseries", PJ segfaults.

# Steps to reproduce
- Open PJ and load a ulg file
- create a custom timeseries, e.g. with "Source data" `actuator_controls_0/control.00` and leave the default function `return value*2`
- give it a name and click "Create new time series", add it to a plot. Everything works as it should
- Edit the custom timeseries
- add another curve by going to the "Curve selector" tab and double clicking e.g. `battery_status/average_current_a`. You can insert it before or after the `return ...` line. You do not have to try to use the variable
- click "Modify timeseries"
- PJ segfaults with the following output:
```
Stack trace (most recent call last):
#31   Object "/tmp/.mount_PlotJuXcnTmr/lib/libQt5Core.so.5", at 0x7f8e68f5aa0e, in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#30   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f8e6649249b, in g_main_context_iteration
#29   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f8e664923ef, in g_main_context_dispatch
#28   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f8e66492196, in g_main_context_dispatch
#27   Object "/tmp/.mount_PlotJuXcnTmr/plugins/platforms/../../lib/libQt5XcbQpa.so.5", at 0x7f8e5e3d19df, in QXcbGlIntegrationPlugin::qt_metacall(QMetaObject::Call, int, void**)
#26   Object "/tmp/.mount_PlotJuXcnTmr/lib/libQt5Gui.so.5", at 0x7f8e692ddd1a, in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#25   Object "/tmp/.mount_PlotJuXcnTmr/lib/libQt5Gui.so.5", at 0x7f8e69304a44, in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*)
#24   Object "/tmp/.mount_PlotJuXcnTmr/lib/libQt5Gui.so.5", at 0x7f8e69302b0f, in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*)
#23   Object "/tmp/.mount_PlotJuXcnTmr/lib/libQt5Core.so.5", at 0x7f8e68f05117, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#22   Object "/tmp/.mount_PlotJuXcnTmr/lib/libQt5Widgets.so.5", at 0x7f8e698b5056, in QApplication::notify(QObject*, QEvent*)
#21   Object "/tmp/.mount_PlotJuXcnTmr/lib/libQt5Widgets.so.5", at 0x7f8e698ad64b, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#20   Object "/tmp/.mount_PlotJuXcnTmr/lib/libQt5Widgets.so.5", at 0x7f8e69909442, in QDesktopWidget::qt_metacall(QMetaObject::Call, int, void**)
#19   Object "/tmp/.mount_PlotJuXcnTmr/lib/libQt5Widgets.so.5", at 0x7f8e699069e5, in QDesktopWidget::qt_metacall(QMetaObject::Call, int, void**)
#18   Object "/tmp/.mount_PlotJuXcnTmr/lib/libQt5Widgets.so.5", at 0x7f8e698b461e, in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool)
#17   Object "/tmp/.mount_PlotJuXcnTmr/lib/libQt5Core.so.5", at 0x7f8e68f05117, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#16   Object "/tmp/.mount_PlotJuXcnTmr/lib/libQt5Widgets.so.5", at 0x7f8e698b5c7a, in QApplication::notify(QObject*, QEvent*)
#15   Object "/tmp/.mount_PlotJuXcnTmr/lib/libQt5Widgets.so.5", at 0x7f8e698ad64b, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#14   Object "/tmp/.mount_PlotJuXcnTmr/lib/libQt5Widgets.so.5", at 0x7f8e698ec677, in QWidget::event(QEvent*)
#13   Object "/tmp/.mount_PlotJuXcnTmr/lib/libQt5Widgets.so.5", at 0x7f8e699a854c, in QAbstractButton::mouseReleaseEvent(QMouseEvent*)
#12   Object "/tmp/.mount_PlotJuXcnTmr/lib/libQt5Widgets.so.5", at 0x7f8e699a83cd, in QAbstractButton::keyPressEvent(QKeyEvent*)
#11   Object "/tmp/.mount_PlotJuXcnTmr/lib/libQt5Widgets.so.5", at 0x7f8e699a6723, in QAbstractButton::toggled(bool)
#10   Object "/tmp/.mount_PlotJuXcnTmr/lib/libQt5Widgets.so.5", at 0x7f8e699a64f1, in QAbstractButton::clicked(bool)
#9    Object "/tmp/.mount_PlotJuXcnTmr/lib/libQt5Core.so.5", at 0x7f8e68f31743, in QMetaObject::activate(QObject*, int, int, void**)
[1]    27110 segmentation fault (core dumped)  PlotJuggler-2.1.8-x86_64.AppImage
```

This is with PJ 2.1.8.

Here is a log with which to reproduce:
[log_2019-03-27-00-34-42.ulg.zip](https://github.com/facontidavide/PlotJuggler/files/3022388/log_2019-03-27-00-34-42.ulg.zip)

facontidavide, 2019-03-29:
the archive you send me seems to be corrupted (fails  to unzip)

ndepal, 2019-03-29:
Try this: https://drive.google.com/file/d/1EB0PAKmGvgm-YGOi05yFqWkits7vVQWv/view?usp=sharing

ndepal, 2019-03-29:
It could not reproduce this with a ROS bag either.

facontidavide, 2019-03-29:
Ok, now I can reproduce it :)
Working on it...

facontidavide, 2019-03-29:
WOW you find a CRITICAL bug. Thanks you so much for reporting.

1042e51229bbe0df568722820a56a03356b60c98

closed 2019-03-29 by facontidavide

ndepal, 2019-03-29:
Happy to help. Thanks for the blazing fast response!

atyshka, 2019-06-18:
Thanks for the fix, will this be added to the ros buildfarm soon so we don't have to build from source?