PlotJuggler/PlotJuggler #169: (Manual) offset removal in XY-plots

llvs, 2019-04-01:
Dear Davide,

thanks for your ongoing effort with PlotJuggler!

Now with equal axis scaling working nicely in XY plots (thank you!), my next issue is that XY plots may feature very large values (e.g. UTM coordinates) where the axis label changes to scientific notation which makes it hard to judge the plot, especially with changes of the lower or fractional digits.

It would be very handy to be able to subtract the starting point as an offset either automatically or manually. Then, I could easily see the traveled distance of the robot.

I tried to apply a custom function which removes the offset but this seems to break the XY-plot and the plot then reverts to a normal plot.

Best regards
llvm

facontidavide, 2019-04-01:
Interesting... the custom function should work in theory to remove the offset.
Can you please share the rosbag to be able to reproduce this and see if I can see anything about it?

marceloparavisi, 2019-10-03:
Hello

I added this functionality in my forked repository (https://github.com/marceloparavisi/PlotJuggler). 

Regards

facontidavide, 2019-10-03:
The correct solution to remove offset in this very particular case is using Custom TimeSeries.

@llvs , You mentioned that the manual offset breaks the XY plot and I suggested to send me a rosbag to reproduce the issue, but you didn't... :(

llvs, 2019-10-07:
I know. It is still on my todo list. However, I have been working on other projects since. It is not forgotten :-)

facontidavide, 2021-09-19:
The best solution is to create a custom plot. If anybody needs a more elegant solution: https://github.com/facontidavide/PlotJuggler/discussions/520

closed 2021-09-19 by facontidavide