PlotJuggler/PlotJuggler #170: ULog: Error loading definitions

yutaoliu, 2019-04-10:
The plugin [DataLoad ULog] throw the exception.
Running the binary from windows 10 installation.

facontidavide, 2019-04-17:
I should update the windows binary to the latest version.

Can you please share a ULOG file that I can use to reproduce the issue?

facontidavide, 2019-05-31:
Since you haven't answered, I am closing this issue.

closed 2019-05-31 by facontidavide

CornerOfSkyline, 2019-08-19:
[demo.zip](https://github.com/facontidavide/PlotJuggler/files/3514349/demo.zip)
@facontidavide this is a ulog file. The problem is still on windows 10.

copterspace, 2020-01-31:
Same problem on Windows 10 x64, just tried, here's log file.
[image]

[log_76_2019-11-14-12-11-44.zip](https://github.com/facontidavide/PlotJuggler/files/4138470/log_76_2019-11-14-12-11-44.zip)

facontidavide, 2020-01-31:
even if I am not so much into ULog file support, I will try to have a look at this... one of these days ;)

ayberkozgur, 2020-07-27:
Same problem persisting on 2.8.2.

facontidavide, 2020-07-27:
I am currently focused on plotjuggler 3.0

If people want / need to prioritize this feature or others, consider becoming a sponsor ☺️

facontidavide, 2020-08-14:
I am glad to say that the problem has been fixed

vserluco, 2022-11-23:
I am having this same issue today. Not sure if something broke

commit 5b01a405 (2020-08-14): fix  #170 : problem with ULOG parser in Windows