PlotJuggler/PlotJuggler #171: Plot Axis Don't Render on OSX

keenanjohnson, 2019-05-14:
[image]

keenanjohnson, 2019-05-14:
I built Plot Juggler for OSX and I'm using it to render ulog files. However, it doesn't seem to render the plot axis haha. See my picture above.

Perhaps this is a font issue or something? Any ideas?

keenanjohnson, 2019-05-14:
Here's my std out:

Keenans-MacBook-Pro-2:~ keenanjohnson$ /Users/keenanjohnson/Desktop/PlotJuggler-master/bin/PlotJuggler ; exit;
"libDataLoadCSV.dylib" : is a DataLoader plugin
"libDataLoadULog.dylib" : is a DataLoader plugin
"libDataStreamSample.dylib" : is a DataStreamer plugin
"libDataStreamSample.dylib" ...but will be ignored unless the argument -t is used.
"libDataStreamWebSocket.dylib" : is a DataStreamer plugin
objc[49271]: Class FIFinderSyncExtensionHost is implemented in both /System/Library/PrivateFrameworks/FinderKit.framework/Versions/A/FinderKit (0x7fff988d53c8) and /System/Library/PrivateFrameworks/FileProvider.framework/OverrideBundles/FinderSyncCollaborationFileProviderOverride.bundle/Contents/MacOS/FinderSyncCollaborationFileProviderOverride (0x1139b0f50). One of the two will be used. Which one is undefined.

facontidavide, 2019-05-15:
Sorry, I am afraid that I have no clue and I can hardly debug it, since I don't have OSX....

If anyone has any clue, please contribute.

keenanjohnson, 2019-05-15:
Any tips on how to debug it? I’m happy to help, just need a few pointers

facontidavide, 2019-05-15:
I honestly have no clue. I am using QWT https://qwt.sourceforge.io/

You might want to download it and check if the examples work correctly or not.

keenanjohnson, 2019-05-15:
Ok thanks very much! I'm going to poke around and see if i can discover the issue.

facontidavide, 2019-10-09:
I will never support OSX, so I am temporarily closing this issue.
If anyone can submit a fix, he/she is welcome.

closed 2019-10-09 by facontidavide