PlotJuggler/PlotJuggler #177: Point series data synchonisation

Guillaumebeuzeboc, 2019-05-31:
Hello,

I have been trying to work with point series (in my case with a custom plugin).
Point series case made me face some trouble since my data are not perfectly synchronized.
First I noticed:
https://github.com/facontidavide/PlotJuggler/blob/master/plotter_gui/point_series_xy.cpp#L54-L56
Axis sizes doesn't seems to be check correctly since `_x_axis->size()` is compared to itself. I would have expected:
```
const size_t data_size =  _x_axis->size();

if( data_size != _y_axis->size() )
``` 

I noticed this since my data are not always synchronised. To explain the cases I am facing I will take an example. Let say I want to display A and B and both are generated with a timestamp (from the original soft sending the data to the plugin). A and B are generated one right after the other but of course have a different timestamp :
- `updateCache()` is called after the plugin received 3xA and 2xB my axes are not the same size but it's just because the last B value is comming
- When `updateCache()` is called there is no A and B that have the exact same timestamp since they are generated one after the other (the difference of timestamp could be something like 100 ms).

So my question is: Is there a way to synchonise the data for point series? Or should the `updateCache()` method be improved to manage these cases?
I went for the second (but dirty version) [solution](https://github.com/Guillaumebeuzeboc/PlotJuggler/commit/df4ec2fba4fab8fd8d85b8dd4649c26ba865a26b), but I am pretty sure that someone else already faced such issue so that is why I am posting here (but maybe something could be improved in PlotJuggler).

facontidavide, 2019-05-31:
Hi,

nice catch, i just corrected the bug.

Unfortunately there is no easy way to solve your use case (and similar one).

We can not tolerate even slight difference in timestamps, otherwise it is not clear how to create a match between X and Y values.
100 ms in many applications (probably not yours) is HUGE.

> So my question is: Is there a way to synchonise the data for point series? Or should the updateCache() method be improved to manage these cases?

If we allow different an X-Y pair to be created from different timestamp, our data become inconsistent.
Wring data is even worse than no data, in my opinion.

Anyway, I don't think it is too bad to apply the fix you suggest, i.,e.

       const size_t data_size =  std::min(_x_axis->size(), _y_axis->size());

I applied a couple of changes, but right now I don't think there is way to mix X and Y without linear interpolation, given the timestamps.

That is not on my TODO list for the time being, since I am planning to make a refactoring of the whole XY plot system... this summer.

closed 2019-05-31 by facontidavide

Guillaumebeuzeboc, 2019-05-31:
Thank you for the clarification (and PlotJuggler^^)