PlotJuggler/PlotJuggler #187: Topic name horizontal scrollbar doesn't work within column 1

PaulBouchier, 2019-08-04:
Hi. I am using the latest release, 2.30 of plotjuggler, and the horizontal scrollbar is disabled (shows full width) when I hide column 2. I can't expand the topic list pane enough to see the full topic name, and I can't scroll across the topic name, so with topic names that have long path names the topic name cannot be seen. Is there a workaround?

Thanks

Paul
[image]

=========================
Thanks for contributing to PlotJuggler. You are great!

Please allow me to better understand your problem providing the following information:

- Please use always the master branch or the latest tag, if you can. 

- Check if the problem can be reproduced using the dummy data created by the command line argument "-t" or one of the files in the folder "datasamples".

- If it can't be reproduced with the dummy data, __please__ share the CSV file or the rosbag that can be used to reproduce the problem.

[image]

facontidavide, 2019-08-04:
Ok, I will check it out

facontidavide, 2019-08-05:
Fixed

closed 2019-08-05 by facontidavide

PaulBouchier, 2019-08-05:
Thanks for the quick fix David - this program is now totally awesome and usable and I will demo it to my workmates. Thanks for your good work. I sent a contribution via paypal in appreciation.

facontidavide, 2019-08-05:
[image]