PlotJuggler/PlotJuggler #188: Unable to locate package

jolau, 2019-08-05:
I've tried to install PlotJuggler with
`sudo apt-get install ros-kinetic-plotjuggler `
but I'm getting the error:
```
Reading package lists... Done
Building dependency tree... 0%
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-kinetic-plotjuggler
```
I've did apt update and also updated the GPG keys (https://discourse.ros.org/t/new-gpg-keys-deployed-for-packages-ros-org/9454)

facontidavide, 2019-08-05:
Duplicate #186

closed 2019-08-05 by facontidavide

khaiyichin, 2020-01-02:
I don't think this is fixed. Like @jolau, I updated the GPG keys and did apt update, but I still get `E: Unable to locate package ros-kinetic-plotjuggler`. #186 only addresses if the GPG key is outdated, but not if the key is updated and the package still cannot be found. Any ideas? Or is building from source the only way?