PlotJuggler/PlotJuggler #189: Unable to find Add Prefix

senoa95, 2019-08-05:
I am currently using Version 2.3.0 and I am having trouble adding a prefix (append data) to my current plots. The Data Loader does not give me an option to add prefix as older versions of plot juggler would. 
[image]

facontidavide, 2019-08-05:
Hi,

bad news: the feature has been intentionally removed.

good news: you probably don't need it anymore.

In the past the "add prefix" was used to load multiple rosbags at once and avoid overwriting the old timeseries with the same name.

In version 2.3+, you can select multiple rosbags at the same time in the "Open file..." dialog and the prefix will be added automatically.

senoa95, 2019-08-05:
Awesome! Thank you for promptly answering my question!

closed 2019-08-05 by senoa95