PlotJuggler/PlotJuggler #191: Map topic

lucabonamini, 2019-08-14:
Hi! First of all, thank you for the amazing work!

I want to know if there is the possibility to plot a occupancy grid, or maybe an array of markers, in order to evaluate a vehicle trajectory against a known track.

Cheers,

Luca

facontidavide, 2019-08-15:
Hi Luca,

 PlotJuggler is not meant to display natively any 2D data such as images or occupancy grids.
You may still create your own plugin that take that information and display it samewhere else (another window), but it make no sense, since you can use RViz for that, that is nacily integrated with PJ.

About the array of markers or, more in general, path built from different waypoints, that might be a little easier, but, once again, you need to either convert the original message to a different format that is more "PlotJuggler-friendly" or build a parsing plugin for your message.

So, I am afraid that I can't help you :(

lucabonamini, 2019-08-16:
Thank you Davide, I'll try to write a parser for array messages. 

Regarding RViz compatibility, is it possible to have, for example, a velocity plot on PJ, select a specific point on that, and then see the XY position on the map loaded in RViz?

facontidavide, 2019-08-16:
Not exactly. You can republish entire topics, that RViz will display as usual.

https://vimeo.com/312135435

closed 2019-08-16 by facontidavide

lucabonamini, 2019-08-22:
Forgive me, I could not answer until today. I'll try RViz compatibility asap!

About the array or occupancy grid parser, what parts of the code should I look to try to create my plugin?