PlotJuggler/PlotJuggler #193: x-y axis of XY-plot may switch undesirably

vinbo, 2019-08-19:
when I draw a XY-plot, sometimes it turns out draws a YX-plot. I swear I checked out the order again and again. the terminal gives the info: 
QwtLinearScaleEngine::divideScale: overflow
QwtLinearScaleEngine::divideScale: overflow
QwtLinearScaleEngine::divideScale: overflow
QwtLinearScaleEngine::divideScale: overflow
QwtLinearScaleEngine::divideScale: overflow

facontidavide, 2019-08-21:
Hi,

as said in the template for bug report, please send to me:

- A rosbag/file to reproduce the error (better if it can be reproduced with the dummy data of option -t)
- Either a screencast of how to reproduce the error or a step by step description.

Cheers

facontidavide, 2019-09-11:
Hi, I would like to solve this but I need you to provide more information about how to reproduce this