PlotJuggler/PlotJuggler #194: UDPROS support

alexswerner, 2019-08-19:
Hi,

it would be nice to have optional UDP support in the DataStreamROS plugin. This would make plotjuggler more error tolerance/ more feasible on laggy wifi connections. Theoretically this should be as simple as replacing:
        ops.transport_hints = ros::TransportHints().tcpNoDelay();
with
        ops.transport_hints = ros::TransportHints().udp();

However this does not work for some reason which seems to be related to plotjuggler. No callback is activated after this change. If i create a minimal UDPROS subscriber this works. Do you have a clue why this could be happening?

facontidavide, 2019-08-21:
I have never used the UDP transport and I don't know why this is happening.

I don't have much time right now, but I will have a look at this in September.

facontidavide, 2019-10-16:
I have no idea of why this is happening. 

Furthermore, I am not sure it is desirable, because when you are receiving data, you don't want to loose messages, because otherwise your interpretation of what is happening may be wrong.

Closing the issue.

closed 2019-10-16 by facontidavide