PlotJuggler/PlotJuggler #196: Deterministic line color sequence

alexswerner, 2019-08-23:
Hi,

it would be nice to have deterministic colors when adding lines to a plot. Whenever i create a new tab with a single plot and add multiple lines, the order of colors seems to be different. I suggest that the order in which lines are added to the plot determines the colors. Maybe the Light/Dark mode should predefine a set of colors.

Cheers,
Alex

JimmyDaSilva, 2019-08-23:
Good point @alexswerner.
Also, many times, I had to change the "random" color of a new added curve because it was hardly visible (light gray) over the white background.
The light/dark mode should set a palette of possible randoms color.

Thanks @facontidavide. I am using this great plotter on a daily basis now.

facontidavide, 2019-08-23:
I have been waiting people to ask this for a long time ;)

Actually, once you drag a new timeseries into a plot, a color is assigned to it and is reused if the same timeseries is added to another plot.

In other words...
[image]

What I do is to give the color blue to timeseries `pose_x` once it is dropped for the first time. Every time you use `pose_x` afterword, it will be blue.

@JimmyDaSilva, I improved a bit the palette in 2.3.0, but I need to create different palettes for dark and light theme, because the default one will not have sufficient contrast for both themes.

If anyone wants to suggest a palette, feel free to propose a PR modifying this file: https://github.com/facontidavide/PlotJuggler/blob/master/include/PlotJuggler/random_color.h

For reference:

https://seaborn.pydata.org/tutorial/color_palettes.html

https://ux.stackexchange.com/questions/85088/ideal-color-palette-for-24-colors

http://repec.sowi.unibe.ch/stata/palettes/getting-started.html

facontidavide, 2019-09-30:
I improved the palette of colors and it should be less annoying.

Since I believe the current behavior is the "right one", I will close the issue. But thanks for the feedback :+1:

closed 2019-09-30 by facontidavide

facontidavide, 2019-12-18:
Please test this branch https://github.com/facontidavide/PlotJuggler/tree/curve_colors