PlotJuggler/PlotJuggler #198: Displaying a bool message

benhirsig, 2019-09-10:
Hello Davide,

I am a new user to PlotJuggler and am using it for plotting some data collected from my robot.

I have a few messages being published whereby the message type is a bool (signifying the power on/off of an electromagnet). I'm trying to plot the change in state against the torque required to run the wheels to see a spike. However, when plotting these bool type messages, I can only plot a line with a gradient, and not a sharp spike similar to that seen in a square wave.

[image]

I was wondering if there is a functionality to correctly display the bool as either a constant 1 or 0. Or if there are plans to include this in the works!

Cheers, Ben

facontidavide, 2019-09-10:
This is expected behavior. If it is easier for you, you can visualize the bool plot as points instead of lines.

For the time being, I am not planning to modify this in any way.

benhirsig, 2019-09-10:
Thanks for your swift response!

I understand that this won't get changed, but for future users, I came up with a workaround.

In ROS, when I intended to change it to False, I passed the True message followed instantly by the False message, and vice versa. This allowed for the sharp (almost 90 degree) lines slopes on the graph.

[image]

Cheers, Ben

facontidavide, 2019-09-11:
ok, I am glad you find a workaround. Cheers

closed 2019-09-11 by facontidavide