PlotJuggler/PlotJuggler #199: plotjuggler crashes

bysteven, 2019-09-10:
Playing a bag both in streaming mode or loading it in plotjuggler, I get the error message pasted below. A bag to reproduce can be downloaded here: https://drive.google.com/file/d/12DhLr8q9NU3KNsAtYtoOOXMvpMBTUr6D/view?usp=sharing

`QwtLinearScaleEngine::divideScale: overflow
QwtLinearScaleEngine::divideScale: overflow
terminate called after throwing an instance of 'std::runtime_error'
  what():  buildRosFlatType: There was an error parsing the buffer
Stack trace (most recent call last) in thread 10402:
#18   Object "", at 0xffffffffffffffff, in 
#17   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f0bc7b1a41c, in clone
      Source "/build/glibc-Cl5G7W/glibc-2.23/misc/../sysdeps/unix/sysv/linux/x86_64/clone.S", line 109, in  [0x7f0bc7b1a41c]
#16   Object "/lib/x86_64-linux-gnu/libpthread.so.0", at 0x7f0bc76a36b9, in start_thread
#15   Object "/usr/lib/x86_64-linux-gnu/libboost_thread.so.1.58.0", at 0x7f0bac1be5d4, in boost::this_thread::interruption_point()
#14   Object "/opt/ros/kinetic/lib/libroscpp.so", at 0x7f0badb1f343, in ros::AsyncSpinnerImpl::threadFunc()
#13   Object "/opt/ros/kinetic/lib/libroscpp.so", at 0x7f0badac30fa, in ros::CallbackQueue::callAvailable(ros::WallDuration)
#12   Object "/opt/ros/kinetic/lib/libroscpp.so", at 0x7f0badac16f7, in ros::CallbackQueue::callOneCB(ros::CallbackQueue::TLS*)
#11   Object "/opt/ros/kinetic/lib/libroscpp.so", at 0x7f0badb1be2c, in ros::SubscriptionQueue::call()
#10   Object "/home/steffen/projects/ws_plotjuggler/devel/lib/plotjuggler/libDataStreamROS.so", at 0x7f0b9e6edd70, in ros::SubscriptionCallbackHelperT<boost::shared_ptr<topic_tools::ShapeShifter const> const&, void>::call(ros::SubscriptionCallbackHelperCallParams&)
    | Source "/opt/ros/kinetic/include/ros/subscription_callback_helper.h", line 144, in ros::SubscriptionCallbackHelperT<boost::shared_ptr<topic_tools::ShapeShifter const> const&, void>::call(ros::SubscriptionCallbackHelperCallParams&)
    |   142:   {
    |   143:     Event event(params.event, create_);
    | > 144:     callback_(ParameterAdapter<P>::getParameter(event));
    |   145:   }
      Source "/usr/include/boost/function/function_template.hpp", line 773, in boost::function1<void, boost::shared_ptr<topic_tools::ShapeShifter const> >::operator()(boost::shared_ptr<topic_tools::ShapeShifter const>) const [0x7f0b9e6edd70]
        770:         boost::throw_exception(bad_function_call());
        771: 
        772:       return get_vtable()->invoker
      > 773:                (this->functor BOOST_FUNCTION_COMMA BOOST_FUNCTION_ARGS);
        774:     }
        775: 
        776:     // The distinction between when to use BOOST_FUNCTION_FUNCTION and
#9    Object "/home/steffen/projects/ws_plotjuggler/devel/lib/plotjuggler/libDataStreamROS.so", at 0x7f0b9e6e4641, in boost::detail::function::void_function_obj_invoker1<boost::function<void (boost::shared_ptr<topic_tools::ShapeShifter const> const&)>, void, boost:
:shared_ptr<topic_tools::ShapeShifter const> >::invoke(boost::detail::function::function_buffer&, boost::shared_ptr<topic_tools::ShapeShifter const>)
    | Source "/usr/include/boost/function/function_template.hpp", line 159, in boost::detail::function::void_function_obj_invoker1<boost::function<void (boost::shared_ptr<topic_tools::ShapeShifter const> const&)>, void, boost::shared_ptr<topic_tools::ShapeShifter c
onst> >::invoke(boost::detail::function::function_buffer&, boost::shared_ptr<topic_tools::ShapeShifter const>)
    |   157:           else
    |   158:             f = reinterpret_cast<FunctionObj*>(function_obj_ptr.obj_ptr);
    | > 159:           BOOST_FUNCTION_RETURN((*f)(BOOST_FUNCTION_ARGS));
    |   160:         }
    |   161:       };
      Source "/usr/include/boost/function/function_template.hpp", line 773, in boost::function1<void, boost::shared_ptr<topic_tools::ShapeShifter const> const&>::operator()(boost::shared_ptr<topic_tools::ShapeShifter const> const&) const [0x7f0b9e6e4641]
        770:         boost::throw_exception(bad_function_call());
        771: 
        772:       return get_vtable()->invoker
      > 773:                (this->functor BOOST_FUNCTION_COMMA BOOST_FUNCTION_ARGS);
        774:     }
        775: 
        776:     // The distinction between when to use BOOST_FUNCTION_FUNCTION and
#8    Object "/home/steffen/projects/ws_plotjuggler/devel/lib/plotjuggler/libDataStreamROS.so", at 0x7f0b9e6d6587, in DataStreamROS::topicCallback(boost::shared_ptr<topic_tools::ShapeShifter const> const&, std::__cxx11::basic_string<char, std::char_traits<char>, st
d::allocator<char> > const&)
      Source "/home/steffen/projects/ws_plotjuggler/src/PlotJuggler/plugins/ROS/DataStreamROS/datastream_ROS.cpp", line 94, in DataStreamROS::topicCallback(boost::shared_ptr<topic_tools::ShapeShifter const> const&, std::__cxx11::basic_string<char, std::char_traits<
char>, std::allocator<char> > const&) [0x7f0b9e6d6587]
         91:     _prev_clock_time = msg_time;
         92: 
         93:     MessageRef buffer_view( buffer );
      >  94:     _ros_parser.pushMessageRef( topic_name, buffer_view, msg_time );
         95: 
         96:     std::lock_guard<std::mutex> lock( mutex() );
         97:     const std::string prefixed_topic_name = _prefix + topic_name;
#7    Object "/home/steffen/projects/ws_plotjuggler/devel/lib/plotjuggler/libDataStreamROS.so", at 0x7f0b9e6fcc98, in RosMessageParser::pushMessageRef(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, MessageRef const&, double)
      Source "/home/steffen/projects/ws_plotjuggler/src/PlotJuggler/plugins/ROS/RosMsgParsers/ros_parser.cpp", line 126, in RosMessageParser::pushMessageRef(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, MessageRef const&, d
ouble) [0x7f0b9e6fcc98]
        123:                 topic_name,
        124:                 msg_view,
        125:                 &flat_container,
      > 126:                 _max_array_size );
        127: 
        128:     if( !max_size_ok && _warnings_enabled )
        129:     {
#6    Object "/home/steffen/projects/ws_plotjuggler/devel/lib/libros_type_introspection.so", at 0x7f0bade126e4, in RosIntrospection::Parser::deserializeIntoFlatContainer(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, absl::S
pan<unsigned char>, RosIntrospection::FlatMessage*, unsigned int) const
#5    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f0bc8389918, in __cxa_throw
#4    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f0bc8389700, in std::terminate()
#3    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f0bc83896b5, in std::rethrow_exception(std::__exception_ptr::exception_ptr)
#2    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f0bc838b84c, in __gnu_cxx::__verbose_terminate_handler()
#1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f0bc7a4a029, in abort
      Source "/build/glibc-Cl5G7W/glibc-2.23/stdlib/abort.c", line 89, in __GI_abort [0x7f0bc7a4a029]
#0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f0bc7a48428, in gsignal
      Source "/build/glibc-Cl5G7W/glibc-2.23/signal/../sysdeps/unix/sysv/linux/raise.c", line 54, in __GI_raise [0x7f0bc7a48428]
Aborted (Signal sent by tkill() 9889 1000)
Aborted (core dumped)
`

facontidavide, 2019-09-11:
Hi,

I don't have permission to download the file. Additionally, I need to know which topic is causing the error.

Cheers

facontidavide, 2019-09-16:
Close because inactivity

closed 2019-09-16 by facontidavide