PlotJuggler/PlotJuggler #202: Not fetching timestamp for x axis?

msadowski, 2019-09-20:
Hi Davide!

Thanks for making PlotJuggler, I'm a big fan! I'm using the released version of PlotJuggler so apologies if it has been fixed already.

Recently when playing back some data from bag files I've noticed what seems like time series not being loaded from series, resulting in plots like the one on the top in this screenshot:

[image]

Basically when plotting the /rr_robot/mobile_base_controller/odom* fields that time starts at 0, while when I plot /rs_t265/imu/* fields the time used is the one from timestamp.

Looking at the value of timestamps in both messages they seem fine:
[image]

Here is the bag that I used.
[bag_filtered.zip](https://github.com/facontidavide/PlotJuggler/files/3634961/bag_filtered.zip)

facontidavide, 2019-09-20:
Interesting. I will have a look

facontidavide, 2019-09-20:
I am not able to reproduce the error using the latest version on master not binary version 2.3.0 (kinetics)

[image]

msadowski, 2019-09-21:
Hi Davide!

This is interesting. I'm on Ubuntu 18.04 and running ROS Melodic. I just built the latest master and I was still able to reproduce it. I've recorded my screen while reproducing this in case it's useful for debugging: https://youtu.be/mFuenFWB0k8.

I'm happy to place some debug statements in the code to get to the source of the issue, if you have any hints where to start then let me know!

facontidavide, 2019-09-21:
That was a tough bug, that was visible only in streaming. Thanks for reporting

msadowski, 2019-09-21:
That was quick! Thanks!!

I just tested it and it works like a charm now!

commit 6be32ff3 (2019-09-21): Fix #202 use_header_stamp not initialized for built-in types