PlotJuggler/PlotJuggler #210: window 10 - menus are slow

bizulk, 2019-10-03:
Hello !

Using version "windows installer 2.1.4" 

When calling a load of a csv file, the menus are really slow to appear.
The plot use is really fast compared to it.

Here are the log startup : 
```
"D3Dcompiler_47.dll" :  "Failed to extract plugin meta data from 'C:/Program Files (x86)/PlotJuggler/D3Dcompiler_47.dll'"
"libDataLoadCSV.dll" : is a DataLoader plugin
"libDataLoadULog.dll" : is a DataLoader plugin
"libDataStreamSample.dll" : is a DataStreamer plugin
"libDataStreamSample.dll" ...but will be ignored unless the argument -t is used.
"libDataStreamWebSocket.dll" : is a DataStreamer plugin
"libEGL.dll" :  "Failed to extract plugin meta data from 'C:/Program Files (x86)/PlotJuggler/libEGL.dll'"
"libgcc_s_seh-1.dll" :  "Failed to extract plugin meta data from 'C:/Program Files (x86)/PlotJuggler/libgcc_s_seh-1.dll'"
"libGLESV2.dll" :  "Failed to extract plugin meta data from 'C:/Program Files (x86)/PlotJuggler/libGLESV2.dll'"
"libstdc++-6.dll" :  "Failed to extract plugin meta data from 'C:/Program Files (x86)/PlotJuggler/libstdc++-6.dll'"
"libwinpthread-1.dll" :  "Failed to extract plugin meta data from 'C:/Program Files (x86)/PlotJuggler/libwinpthread-1.dll'"
"opengl32sw.dll" :  "Failed to extract plugin meta data from 'C:/Program Files (x86)/PlotJuggler/opengl32sw.dll'"
"Qt5Core.dll" :  "Failed to extract plugin meta data from 'C:/Program Files (x86)/PlotJuggler/Qt5Core.dll'"
"Qt5Gui.dll" :  "Failed to extract plugin meta data from 'C:/Program Files (x86)/PlotJuggler/Qt5Gui.dll'"
"Qt5Network.dll" :  "Failed to extract plugin meta data from 'C:/Program Files (x86)/PlotJuggler/Qt5Network.dll'"
"Qt5Qml.dll" :  "Failed to extract plugin meta data from 'C:/Program Files (x86)/PlotJuggler/Qt5Qml.dll'"
"Qt5Quick.dll" :  "Failed to extract plugin meta data from 'C:/Program Files (x86)/PlotJuggler/Qt5Quick.dll'"
"Qt5Svg.dll" :  "Failed to extract plugin meta data from 'C:/Program Files (x86)/PlotJuggler/Qt5Svg.dll'"
"Qt5WebSockets.dll" :  "Failed to extract plugin meta data from 'C:/Program Files (x86)/PlotJuggler/Qt5WebSockets.dll'"
"Qt5Widgets.dll" :  "Failed to extract plugin meta data from 'C:/Program Files (x86)/PlotJuggler/Qt5Widgets.dll'"
"Qt5Xml.dll" :  "Failed to extract plugin meta data from 'C:/Program Files (x86)/PlotJuggler/Qt5Xml.dll'"
```

ww9980, 2019-10-29:
set to true the (infamous) SH_Menu_SloppySubMenus flag will solve the problem.

bizulk, 2019-10-29:
I did not find out where to set such Flag ?
I search for ini/xml files in installation directory, launched the maintenance tool, ask my friend Google.

ww9980, 2019-10-29:
Use a style proxy.

bizulk, 2019-10-30:
I search in the doc where would be such config file, there nothing about that.
In the "styles" installed directory there are only binairies.

linarism, 2019-11-29:
I'm also having trouble figuring this out, I'm not familiar with the tools used to build PlotJuggler

facontidavide, 2020-02-07:
could you please check the latest commit. I think I fixed it

closed 2020-02-07 by facontidavide

bizulk, 2020-02-10:
Hello.
OK I'll do a check !

linarism, 2020-02-11:
I'm not able to compile the Windows version of the recent commit. @facontidavide3 would you have an .exe to test with?