PlotJuggler/PlotJuggler #212: XYPlot inversion when saving layout

ClementAubry, 2019-10-07:
Hi Davide, first of all many thanks for your incredible tool...

I found a minor bug playing bag files (streamer on) : when saving and replacing a layout which contain a swapped XYPlot, the "swapping" operation is done at each save and replace..

To reproduce : when playing a bag file, 
  * Save a layout with two XYPlots, one swapped when creating the graph.
  * kill all your bag playing and plotjuggler
  * Play your bag and use your plotjuggler layout => XYPlot unchanged
  * save the layout another time (no changes needed)
  * kill all your bag playing and plotjuggler
  * Play your bag and use your plotjuggler layout => XYPlot swapped

Thanks

facontidavide, 2019-10-07:
thanks, I will check it out

facontidavide, 2019-10-07:
Are you using the latest version on master? 

I am trying to reproduce it following your instruction but with no success

ClementAubry, 2019-10-07:
I'm using the packaged version (and waiting pkged Meme version! ;) ) :

ros-melodic-plotjuggler/bionic,now 2.3.1-1bionic.20190925.125516 amd64

facontidavide, 2019-10-07:
I am afraid that I am doing something different from you, because with the packaged version in kinetic (2.3.0) I can't reproduce the error. Would you be so kind to send me a video (screencast with SimpleScreenRecorder or similar) to see what I am doing wrong?

facontidavide, 2019-10-14:
Can you please check if the problem was solved in the latest version (2.3.5)?

closed 2019-10-16 by facontidavide