PlotJuggler/PlotJuggler #214: Undefined symbol lookup error when loading ROS bag

cowmanjohn, 2019-10-14:
When attempting to load a rosbag in the prebuilt PlotJuggler package (ver 2.3.4-2xenial-20191004-001357+0000), I am getting the error:

`/opt/ros/kinetic/lib/plotjuggler/PlotJuggler: symbol lookup error: /opt/ros/kinetic/lib/plotjuggler/libDataLoadROS.so: undefined symbol: _ZN16RosIntrospection7ROSTypeC1EN5boost16basic_string_refIcSt11char_traitsIcEEE`

I don't have any bags I can share, but I experience the same error loading any of the bags from https://google-cartographer-ros.readthedocs.io/en/latest/data.html

I do not see the error when loading the dummy data or .csv files.

cowmanjohn, 2019-10-14:
I am able to load ROS bags after doing `sudo apt install ros-kinetic-ros-type-introspection` as mentioned in https://github.com/facontidavide/PlotJuggler/issues/166

closed 2019-10-14 by cowmanjohn

facontidavide, 2019-10-14:
What the hell, I believed that was finally fixed :(

I really don't know how to fix this, sorry.

facontidavide, 2019-10-16:
Fixed in AppImage 2.3.6