PlotJuggler/PlotJuggler #216: Crashing when loading bagfile

rikobrandes, 2019-10-15:
AppImage 2.3.5 running on Ubuntu 16.04 LTS: 

"libDataLoadCSV.so" : is a DataLoader plugin
"libDataLoadROS.so" : is a DataLoader plugin
"libDataLoadULog.so" : is a DataLoader plugin
"libDataStreamROS.so" : is a DataStreamer plugin
"libDataStreamSample.so" : is a DataStreamer plugin
"libDataStreamSample.so" ...but will be ignored unless the argument -t is used.
"libRosoutPublisher.so" : is a StatePublisher plugin
"libRosTopicPublisher.so" : is a StatePublisher plugin
terminate called without an active exception
Stack trace (most recent call last):
#31   Object "/tmp/.mount_plt5aOQYr/lib/libQt5Gui.so.5", at 0x7f4008445d2a, in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*)
#30   Object "/tmp/.mount_plt5aOQYr/lib/libQt5Core.so.5", at 0x7f4007fdcd27, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#29   Object "/tmp/.mount_plt5aOQYr/lib/libQt5Widgets.so.5", at 0x7f4008abfc1f, in QApplication::notify(QObject*, QEvent*)
#28   Object "/tmp/.mount_plt5aOQYr/lib/libQt5Widgets.so.5", at 0x7f4008ab859b, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#27   Object "/tmp/.mount_plt5aOQYr/lib/libQt5Widgets.so.5", at 0x7f4008b160c2, in QDesktopWidget::qt_metacall(QMetaObject::Call, int, void**)
#26   Object "/tmp/.mount_plt5aOQYr/lib/libQt5Widgets.so.5", at 0x7f4008b137e8, in QDesktopWidget::qt_metacall(QMetaObject::Call, int, void**)
#25   Object "/tmp/.mount_plt5aOQYr/lib/libQt5Widgets.so.5", at 0x7f4008abf0c1, in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool)
#24   Object "/tmp/.mount_plt5aOQYr/lib/libQt5Core.so.5", at 0x7f4007fdcd27, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#23   Object "/tmp/.mount_plt5aOQYr/lib/libQt5Widgets.so.5", at 0x7f4008abffbc, in QApplication::notify(QObject*, QEvent*)
#22   Object "/tmp/.mount_plt5aOQYr/lib/libQt5Widgets.so.5", at 0x7f4008ab859b, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#21   Object "/tmp/.mount_plt5aOQYr/lib/libQt5Widgets.so.5", at 0x7f4008c35e02, in QMenu::event(QEvent*)
#20   Object "/tmp/.mount_plt5aOQYr/lib/libQt5Widgets.so.5", at 0x7f4008af8ea7, in QWidget::event(QEvent*)
#19   Object "/tmp/.mount_plt5aOQYr/lib/libQt5Widgets.so.5", at 0x7f4008c3384f, in QMenu::mouseReleaseEvent(QMouseEvent*)
#18   Object "/tmp/.mount_plt5aOQYr/lib/libQt5Widgets.so.5", at 0x7f4008c328e5, in QMenu::leaveEvent(QEvent*)
#17   Object "/tmp/.mount_plt5aOQYr/lib/libQt5Widgets.so.5", at 0x7f4008c2b389, in QMenu::setIcon(QIcon const&)
#16   Object "/tmp/.mount_plt5aOQYr/lib/libQt5Widgets.so.5", at 0x7f4008ab46ef, in QAction::activate(QAction::ActionEvent)
#15   Object "/tmp/.mount_plt5aOQYr/lib/libQt5Widgets.so.5", at 0x7f4008ab20a1, in QAction::triggered(bool)
#14   Object "/tmp/.mount_plt5aOQYr/lib/libQt5Core.so.5", at 0x7f400800a7d3, in QMetaObject::activate(QObject*, int, int, void**)
[1]    30521 segmentation fault (core dumped)

facontidavide, 2019-10-15:
You didn't include the entire message. Can you PLEASE share with me the rosbag to reproduce it, please?

[image]

facontidavide, 2019-10-16:
You might need to upgrade ros_type_introspection

      sudo apt install ros-kinetic-ros-type-introspection

check that you have version 2.02 installed

facontidavide, 2019-10-16:
Additionally, please try the latest [AppImage 2.3.6](https://github.com/facontidavide/PlotJuggler/releases/download/2.3.6/PlotJuggler-2.3.6-x86_64.AppImage)

Closing the issue. Feel free to reopen if none of the solutions I suggested works for you.

facontidavide, 2019-10-16:
Please try the latest AppImage 2.3.6

Closing for the time being. Reopen if the problem persist...

closed 2019-10-16 by facontidavide