PlotJuggler/PlotJuggler #217: Unable to start PlotJuggler-2.3.6-x86_64.AppImage

fivef, 2019-10-17:
I get this error when starting the AppImage on Ubuntu 16.04:
Note: Qt5 and libx11-xcb1 are installed on my system. But should be provided by the AppImage anyway, right?

```
./PlotJuggler-2.3.6-x86_64.AppImage       

This application failed to start because it could not find or load the Qt platform plugin "xcb"
in "".

Reinstalling the application may fix this problem.
Stack trace (most recent call last):
#11   Object "", at 0xffffffffffffffff, in 
#10   Object "./PlotJuggler-2.3.6-x86_64.AppImage", at 0x4836d8, in _start
#9    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fec010ff82f, in __libc_start_main
      Source "/build/glibc-Cl5G7W/glibc-2.23/csu/../csu/libc-start.c", line 291, in __libc_start_main [0x7fec010ff82f]
#8    Object "./PlotJuggler-2.3.6-x86_64.AppImage", at 0x4823f4, in main
#7    Object "/tmp/.mount_PlotJuIs0G29/usr/bin/../lib/libQt5Widgets.so.5", at 0x7fec02984a68, in QApplicationPrivate::init()
#6    Object "/tmp/.mount_PlotJuIs0G29/usr/bin/../lib/libQt5Gui.so.5", at 0x7fec023d059d, in QGuiApplicationPrivate::init()
#5    Object "/tmp/.mount_PlotJuIs0G29/usr/bin/../lib/libQt5Core.so.5", at 0x7fec01fdfeea, in QCoreApplicationPrivate::init()
#4    Object "/tmp/.mount_PlotJuIs0G29/usr/bin/../lib/libQt5Gui.so.5", at 0x7fec023ce8bc, in QGuiApplicationPrivate::createEventDispatcher()
#3    Object "/tmp/.mount_PlotJuIs0G29/usr/bin/../lib/libQt5Gui.so.5", at 0x7fec023ce72e, in QGuiApplicationPrivate::createPlatformIntegration()
#2    Object "/tmp/.mount_PlotJuIs0G29/usr/bin/../lib/libQt5Core.so.5", at 0x7fec01de0cd0, in QMessageLogger::fatal(char const*, ...) const
#1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fec01116029, in abort
      Source "/build/glibc-Cl5G7W/glibc-2.23/stdlib/abort.c", line 89, in __GI_abort [0x7fec01116029]
#0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fec01114428, in gsignal
      Source "/build/glibc-Cl5G7W/glibc-2.23/signal/../sysdeps/unix/sysv/linux/raise.c", line 54, in __GI_raise [0x7fec01114428]
Aborted (Signal sent by tkill() 18122 1000)
[1]    18122 abort      ./PlotJuggler-2.3.6-x86_64.AppImage
```

facontidavide, 2019-10-17:
Ohhh man.. I will have a look, thanks for reporting

facontidavide, 2019-10-17:
would you kindly try do download it again?

probonopd, 2019-10-17:
Have you tried to do it 1:1 like in https://github.com/probonopd/linuxdeployqt#using-linuxdeployqt-with-travis-ci? That usually works for me.

facontidavide, 2019-10-17:
@probonopd  unfortunately I had a problem with **linuxdeployqt** and I switched to **linuxdeploy**

https://github.com/linuxdeploy/linuxdeploy

I will report the issue to you.

facontidavide, 2019-10-17:
I have tried two approaches:

https://github.com/facontidavide/PlotJuggler/releases/download/2.3.6/PlotJuggler-2.3.6-x86_64.A.AppImage

First, generate with **linxdeployqt**, but it still miss **libros_type_introspection.so**

https://github.com/facontidavide/PlotJuggler/releases/download/2.3.6/PlotJuggler-2.3.6-x86_64.B.AppImage

Second, generate with **linxdeploy**, that works on my computer, but... not on yours :(

@fivef , can you try both, or at least the second one?

probonopd, 2019-10-17:
> https://github.com/facontidavide/PlotJuggler/releases/download/2.3.6/PlotJuggler-2.3.6-x86_64.A.AppImage

404 :(

probonopd, 2019-10-17:
Is `libros_type_introspection.so` loaded with `dlopen()`? If so, you need to pass it to linuxdeployqt with `-executable=/path/to/libros_type_introspection.so`

facontidavide, 2019-10-17:
I did

fivef, 2019-10-18:
Version B starts but when I try to load a bag file it crashes with:
```
/tmp/.mount_PlotJuvUE7zv/AppRun.wrapped: symbol lookup error: 
/tmp/.mount_PlotJuvUE7zv/usr/bin/libDataStreamROS.so: undefined symbol:
_ZN16RosIntrospection7ROSTypeC1EN5boost16basic_string_refIcSt11char_traitsIcEEE
```

facontidavide, 2019-10-18:
I am about to give up

facontidavide, 2019-10-18:
I have just:

- launched Ubuntu 16.04 with a live CD and 18.04 in another virtual machine.
- I have NO dependencies installed, no ROS installed, no boost installed.

Download a rosbag and version B, and everything just worked for both versions of Ubuntu.

I removed the other two files and I am keeping only version "B".

fivef, 2019-10-18:
Nice! Probably there is something wrong with my bagfile.

facontidavide, 2019-10-18:
not sure... can you please share the rosbag with me?

facontidavide, 2019-10-19:
I am closing this for the time being, but if anyone experience this problem again, please let me know

facontidavide, 2019-10-19:
@fivef I still think that you should share the rosbag with me, to see if i can reproduce the issue

fivef, 2019-10-29:
Sorry for the delay. I tried it again and now I am not able to reproduce the problem with any of my bagfiles. So everything is fine!

facontidavide, 2019-10-29:
happy to hear that

msadowski, 2019-11-14:
@facontidavide I just updated PlotJuggler to 2.3.6-2 through apt-get to get the Memes but it seems I'm getting the same error as @fivef described. 

```
/opt/ros/melodic/lib/plotjuggler/PlotJuggler: symbol lookup error: /opt/ros/melodic/lib/plotjuggler/libDataStreamROS.so: undefined symbol: _ZN16RosIntrospection7ROSTypeC1EN5boost16basic_string_refIcSt11char_traitsIcEEE
```

In my case the error occurs as soon as I want to start the ROS streamer (after I select topics of interest and press OK). I'm running this on a distributed setup, with my robot being the ROS master.

Yesterday everythin worked correctly and I was able to plot stuff live on my laptop. After I updated PlotJuggler I also updated all packages on the robot, so I can't say just yet if this is because of PlotJuggler update or some ros package updating. 

I'll try to troubleshoot this further and let you know if I find a way to make it work

facontidavide, 2019-11-14:
it is really hard for me to make this damn AppImage work :(
In particular because I can hardly reproduce this.

The best I can say is... either wait for the debian package to be synchronized or compile ros_type_introspection and PlotJuggler from source.

Said that, removing maybe boost::string-ref might help solving the problem. I will work on a potential solution and I will ask you to test it soonish.

facontidavide, 2019-11-14:
have you tried 2.4.1?

https://github.com/facontidavide/PlotJuggler/releases/tag/2.4.1

facontidavide, 2019-11-14:
@msadowski 

version 2.4.1 has been just released. Can you try it?

https://discourse.ros.org/t/new-packages-for-melodic-2019-11-14/11494

msadowski, 2019-11-14:
@facontidavide Interestingly after I upgraded all the packages on my non-master computer everything started working again! I was trying very hard to reproduce it after upgrade but I wasn't able to. I'm really curious to know what would have broken it.

facontidavide, 2019-11-14:
I have an hypothesis (I am not sure).

PlotJuggler depend on **ros_type_introspection** (where the error comes from).

It could be a problem of binary compatibility between an old version or **ros_type_introspection**  that was still in your computer and the one used to generate PlotJuggler.

This would explain why the full upgrade fixed the issue.

closed 2019-11-14 by facontidavide

Jwenda, 2019-11-21:
Thanks @facontidavide , I just upgrade the **ros-type-introspection** and it works now.