PlotJuggler/PlotJuggler #218: Feature - Exporting plots as vector images (svg, eps)

aaravrav142, 2019-10-24:
Hi, 
Thanks for providing this wonderful tool to analyze the bag file. My life got so easy with this tool and it has saved me hours of work. One feature that I feel is lacking is the ability to save the plots as vector images (svg or eps format). Currently only .jpg and .png are available. Since most people would like to further enhance and use the  plots for articles, conferences, presentations, I believe such feature would be very valuable to the community. Export of data from plots would be the best.  

Thanks

facontidavide, 2019-10-24:
This feature was present in the past and it can be easily added.

I had some problems with libqt5svg5 in the past and I "temporarily" removed.
I guess I can check again and see if it works correctly.

facontidavide, 2019-10-24:
Have a look to the last commit and tell me what you think

closed 2019-10-24 by facontidavide

aaravrav142, 2019-10-25:
Hi,
Perfect!!! works like a charm. Thanks a tonne for this.