PlotJuggler/PlotJuggler #221: Switching from Custom to "NO Transform" doesn't work

karliss, 2019-11-01:
Version: 6e351971c53d44a086c14be44bdd3cfc9b3d1bd1

### Reproduction steps
1) open simple.csv
2) Add "A" to plot
3) Create function in function editor that multiplies value by 4
4) Apply it to A and observe change in range
5) In the context menu select "NO transform", observe that function is still applied
6)  In the context menu select "1st Derivateive" observe that custom transform gets removed
7) ) In the context menu select "NO transform", observe that original curves are displayed without any transformation

### Excpected behaviour

Custom transforms are removed in step 5

facontidavide, 2019-11-01:
thanks for reporting. I will have a look

facontidavide, 2019-11-01:
Done!

closed 2019-11-01 by facontidavide