PlotJuggler/PlotJuggler #224: Can't plot fields defined after an array in custom message

philimat, 2019-11-06:
Thanks for contributing to PlotJuggler. You are great!

# If you ignore these instructions, I will ignore your issue

Please allow me to better understand your problem providing the following information:

- Please use always the master branch or the latest tag, if you can. 

- Check if the problem can be reproduced using the dummy data created by the command line argument "-t" or one of the files in the folder "datasamples".

- If it can't be reproduced with the dummy data, __please__ share the CSV file or the rosbag that can be used to reproduce the problem.

[image]

I defined a custom ROS message to view some point cloud statistics:

Header header
uint32 points
float32[] intensity
float32 intensity_ave
float32 centroid_distance

When I went to plot the avaiable fields for this message only the header and points were avaiable for plotting. When I changed the ordering of my message fields to below, the centroid_distance and the intensity_ave became available for plotting.

Header header
uint32 points
float32 intensity_ave
float32 centroid_distance
float32[] intensity

This seems like a bug for availability of data to be plotted being order dependent in the message. I installed using sudo apt-get install ros-melodic-plotjuggler .

facontidavide, 2019-11-06:
Thanks for reporting, I will check that

facontidavide, 2019-11-06:
I guess the array intensity is quite large, am I correct?

philimat, 2019-11-06:
Yes this array can get quite large at times. I'm not looking to plot it, only the other items such as points, intensity_ave and centroid_distance.

facontidavide, 2019-11-06:
The problem is in ros_type_introspection.

This is the fix. 

https://github.com/facontidavide/ros_type_introspection/tree/Fix_do_store

if you can try and tell me if it works, that would be great.

- download  ros_type_introspection in the same catkin workspace as plotjuggler.
-  switch to the branch with the fix
- compile both

facontidavide, 2019-11-07:
If you give me a rosbag, I can verify for you that the problem has been fixed. otherwise you have to verify it

closed 2019-11-07 by facontidavide