PlotJuggler/PlotJuggler #232: How to install newest version

Allenhanbo, 2019-12-03:
I use the command:
sudo apt-get install qtbase5-dev libqt5svg5-dev libqt5multimedia ros-$ROS_DISTRO-ros-type-introspection
mkdir -p ws_plotjuggler/src; cd ws_plotjuggler/src
git clone https://github.com/facontidavide/PlotJuggler.git
cd ..
catkin_make
source devel/setup.bash

to install the newest version of plotjuggler. But I still can not use X-Y plot. I would like to know how to fix this?

closed 2019-12-03 by Allenhanbo

facontidavide, 2019-12-04:
What was the problem?

Allenhanbo, 2019-12-04:
I download the new version:https://github.com/facontidavide/PlotJuggler/releases/tag/2.3.3
and I catkin_make successfully. But I still can not do the X-Y plot using my rosbag file: https://drive.google.com/file/d/1H3Fi3bBRqAFn-4YsiPA1HHGdBsXTB2Xo/view?usp=sharing.
I would like to know if I want to do the X-Y plot, the right way is to drag the topic I would like to plot in order?

Like in the bag I shared, if I want to plot vicon/cf1/cf1/transform/translation/x and  vicon/cf1/cf1/transform/translation/y. I should drag these topics in order in the same window?

Allenhanbo, 2019-12-04:
Oh, I fix the issue! I should drag using right mouse. Thanks!

facontidavide, 2019-12-04:
There are help videos to find this kind of Features