PlotJuggler/PlotJuggler #236: Need matching colors for multiple plots

normanhendrich, 2019-12-18:
Hi there, 

is there a way to enforce consistent colors for multiple plots? I am using PlotJuggler 2.3.4 (Ubuntu 16.04, ros-kinetic-plotjuggler) and can't find any such option.

My use case is to plot robot "joint_states" in neighboring plots, one plot for joint positions, one for joint velocities, and one for joint accelerations. Each plot has one timeseries for each robot joint (in my case, a 6-DOF industrial robot). 

Plotting works fine, but with different colors assigned to each of the plots. This makes it quite hard to identify the positions/velocities/accelerations of one joint, as different colors are assigned by default. Using the same colors for each robot joint in the different plots would help a lot.

Even worse, in some plots I want (need) to overlay both the actual and the desired values, and it would be very nice to assign user defined colors (e.g. dark red for actual position of joint 1 and light red for desired position of joint 1, dark/light green for joint 2, etc.). If I add timeseries to one plot, I often end up with repeated colors for different things (e.g. orange for actual position of joint 3 and orange again for desired position of joint 5)... which makes interpretation a bit difficult.

I read about "Deterministic line color sequence #196" and your decision "wont-fix". Still, I would ask you to reconsider. The current random color sequences are fine, but an option to reset the sequence for each new plot subwindow would be very nice.

"Change color" also works, but selecting matching colors for 12..36 individual curves via the color-picker dialog is not fun.

BTW, While not very beautiful, the default color sequence used in other tools (matlab, matplotlib, gnuplot) like red-green-blue-black-cyan-magenta-yellow etc. would maybe also be worth a consideration? Maybe some option like:  
preferences -> color sequence -> 
[x] random colors     (enabled by default, behaves like current PlotJuggler)
[] random colors, shared (sequence restarts for each new subwindow)
[] fixed sequence  (red-green-blue-...)

Best,
  Norman

- Please use always the master branch or the latest tag, if you can. 
Screenshot is from PlotJuggler 2.3.4 from ros-kinetic-plotjuggler. 

[image]

facontidavide, 2019-12-18:
it used to be like this... until someone complained that they don't want matching colors.

I hate users! I agree with you and will revert this to the original behavior

facontidavide, 2019-12-18:
> Plotting works fine, but with different colors assigned to each of the plots. This makes it quite hard to identify the positions/velocities/accelerations of one joint, as different colors are assigned by default.

This might be solved by the option of "resetting the color sequence" in each plot, right? This would be exactly what suggested in #196 

> Even worse, in some plots I want (need) to overlay both the actual and the desired values, and it would be very nice to assign user defined colors (e.g. dark red for actual position of joint 1 and light red for desired position of joint 1, dark/light green for joint 2, etc.). 

I have no idea how to do such a thing.

> BTW, While not very beautiful, the default color sequence used in other tools (matlab, matplotlib, gnuplot) like red-green-blue-black-cyan-magenta-yellow etc. would maybe also be worth a consideration?

I started from the palette of matplotlib, but I slightly tweaked the colors to make them work for both dark and light background.

SUMMARIZING: I agree to ad this option to preferences (eventually)

facontidavide, 2019-12-18:
Will work on it, it will look like this...
[image]

> "Change color" also works, but selecting matching colors for 12..36 individual curves via the color-picker dialog is not fun.

This reminds me that I should eventually add a set of predefined colors to that dialog...

facontidavide, 2019-12-18:
Please test this branch https://github.com/facontidavide/PlotJuggler/tree/curve_colors

closed 2019-12-19 by facontidavide

facontidavide, 2020-01-04:
Have you tried 2.5.0? Does the change address your issue?