PlotJuggler/PlotJuggler #238: feature request: yaw_degrees for imu quaternion

Timple, 2019-12-23:
Or maybe just any quaternion you encounter?

The `yaw_degrees` feature is very convenient for the odometry so thanks for that already! :)

Timple, 2020-01-21:
Figured out how to do it 'manually' using the quaternionToYaw function and this tutorial:
https://vimeo.com/355538901

Feature request remains though

facontidavide, 2020-01-21:
I need you to be a little more precise...

Based on the current architecture, I can "magically" create RPY data only from a specific ROS message type. In your case is

           sensor_msgs/Imu

right?

Timple, 2020-01-22:
Our current usecase is sensor_msgs/Imu, so having a `yaw_degrees` there as you already do for odometry would be great.

---

But the more awesome and complete implementation would be to 'detect' the [Quaternion](https://docs.ros.org/api/geometry_msgs/html/msg/Quaternion.html) msg even if it resides as part of another message (Imu/Odometry/etc).
That way all messages containing a quaternion can be displayed as follows:
```
.../orientation/x
.../orientation/y
.../orientation/z
.../orientation/w
.../orientation/roll
.../orientation/pitch
.../orientation/yaw
```
Where the latter three would be added for convenience since (most?) humans tend to think in euler angles rather than quaternions 

Or perhaps:
```
.../orientation/roll_deg
.../orientation/pitch_deg
.../orientation/yaw_deg
```

facontidavide, 2020-01-23:
i know this is what you were going to propose... :)

It is a PITA, but doable

Timple, 2020-01-27:
If it was very easy I've would have send a PR instead of a feature request :)

Well if message types cannot be distilled, it might be pragmatically done by checking the fields. 
If (and only if) an x,y,z,w field is present, lets add roll, pitch and yaw.

facontidavide, 2020-01-27:
yes, it can be done introspecting this fields of the message... the parsing library is called **ros_type_introspection** for a reason :)

I just have no bandwidth for the time being to do this, I am horribly busy.

facontidavide, 2020-02-09:
Done.

closed 2020-02-09 by facontidavide

Timple, 2020-02-18:
Works for imu message, thanks! :dancing_men: