PlotJuggler/PlotJuggler #240: core dump

carlin314, 2020-01-15:
First, **great thanks for your guys, plotjugger is really really great**. **You guys are so great**!

My feedback, just a little problem, and I think not a big problem, just a qt issue, I think.
- I use app image ➜ ./PlotJuggler-2.4.1-x86_64.AppImage in ubuntu 16.04
- I use app image in system, then I change some layout, then I just change my mind.I decide to change back to original layout,so I click load recent layout, and it crash with message under.

> QMetaObject::connectSlotsByName: No matching signal for **on_checkBoxCaseSensitive_toggled**(bool)
> QMetaObject::connectSlotsByName: No matching signal for **on_actionFullscreen_triggered**()

[image]

facontidavide, 2020-01-16:
I think the problem is really somewhere else. Using the latest version on master might help.

I must create the AppImage for 2.5.0 ...

facontidavide, 2020-02-08:
Hi,

can you try a new version, 2.5.1.

Can you please provide the steps to reproduce this issue?

facontidavide, 2020-02-09:
I am closing this for the time being. Let me know if it still happens with the newest version and how to reproduce it in that case.

closed 2020-02-09 by facontidavide