PlotJuggler/PlotJuggler #241: Feature idea: MAVLink streaming

mortenfyhn, 2020-01-16:
Hi, and thanks for making plotjuggler. It's super nice.

Has a streaming plugin for MAVLink data ever been considered? It would probably be useful to PX4 and ArduPilot users/developers. I'm considering to implement this myself, but I'd like to check if anyone has looked into this before.

facontidavide, 2020-01-16:
Hi,

even if that would be cool, I am not working anymore with droves, therefore I will definitively not work on this feature, unless it is sponsored.

I think it would be great if you want to do it. I will be happy to give you suggestions about how to write plugins, if you need.

Cheers

mortenfyhn, 2020-01-16:
I understand, thanks for the feedback. I'll contact you if I go for it, and get stuck!

facontidavide, 2020-01-16:
Looking forward for your Pull Request ;)

closed 2020-02-08 by facontidavide

thezenox, 2025-08-28:
> Hi, and thanks for making plotjuggler. It's super nice.
> 
> Has a streaming plugin for MAVLink data ever been considered? It would probably be useful to PX4 and ArduPilot users/developers. I'm considering to implement this myself, but I'd like to check if anyone has looked into this before.

Here you go
https://github.com/Neomium/DataStreamMavLink